nisargjhaveri/WirelessAndroidAutoDongle #183: USB: device is not supported - VW Golf (Mk7)/Pixel 7/Pi02W

parherman, 2024-09-11:
I am unable to connect my Pixel 7 to my VW Golf (Mk7) using Pi Zero 2W (build 4.1.0)

The Pi Zero shows up in the head unit, but it reports: "USB: device is not supported" when trying to connect  

On the phone it looks like Android Auto is starting, but closes down immediately

The notification "Looking for Android Auto" is always showing, but sometimes also "Connected to your car"
 
The phone works over USB

[messages.txt](https://github.com/user-attachments/files/16970289/messages.txt)

parherman, 2024-09-12:
Will try with another phone and try some debugging tips from #176

ryanhoetger, 2024-09-14:
Having issues with my Tiguan as well

parherman, 2024-09-22:
Small update: Tested with another phone, Samsung A50, but get the same result, "USB: device is not supported", from the head unit

ganeshenoy, 2024-09-25:
Hi. Pixel 6 + VW Jetta (MIB2 HU) user here. I was facing the exact same problem. The connection was only made after I clicked on 'AAWirelessDongle' in the devices list on the HU. 

On subsequent attempts, I still get the 'USB: Device is not supported' warning every time the phone connects, but then it goes ahead and connects. Not sure if that can be fixed. 

Hope this helps.

nisargjhaveri, 2024-09-25:
> On subsequent attempts, I still get the 'USB: Device is not supported' warning every time the phone connects, but then it goes ahead and connects after a few seconds. @nisargjhaveri, can this be fixed?

This is even stranger. Any chance you can share the logs?

The "USB: Device is not supported" issue seems to be commonly coming either because of bad usb cable or insufficient power looking at the other reports.

ganeshenoy, 2024-09-26:
Thanks for your reply, @nisargjhaveri. And also for your effort in designing, developing, testing, and supporting this wonderful solution. 

I will try to get the logs this weekend. I was just wondering if changing the connection sequence using the AAWG_CONNECTION_STRATEGY variable could help in this situation. It looks like the program is trying to connect via USB before it is completely ready, if that makes any sense.

nisargjhaveri, 2024-09-26:
> It looks like the program is trying to connect via USB before it is completely ready, if that makes any sense.

If this is the case, different connection strategy could actually help. "Dongle mode" would wait for the headunit BT to connect to phone and only then try to initialize AA connection. "USB first" will wait for the headunit to send usb accessory start signal before initiating connection with phone via BT/Wifi. I'd suggest start with "Dongle mode" and if it still does not work, see if "usb first" helps.

ganeshenoy, 2024-09-26:
Got it. Thanks. Sounds like "dongle mode" may do the trick for me. I just need to open the config file on the SD card with a text editor like Notepad and change the value, right?

ganeshenoy, 2024-09-26:
How do I change the value of the AAWG_CONNECTION_STRATEGY variable, @nisargjhaveri? Will it require me to build my own image file?

parherman, 2024-09-26:
@ganeshenoy I believe you should be able to edit this file: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/aawgd.conf.sh directly on the SD card whiteout building your own image

ganeshenoy, 2024-09-26:
[image]

These are the contents of the SD card I see when I open it in Windows Explorer on my laptop, @parherman. The 'overlays' folder just contains a large number of the DTBO files. I can't find the aawgd config file anywhere...

parherman, 2024-09-26:
Its on the rootfs partition - but it is probably only easily accessible on Linux, sorry!

ganeshenoy, 2024-09-27:
[image]
I tried mounting the SD card using the Linux File Systems for Windows program by Paragon, and the rootfs partition is read-only. The 'Mount in Read/Write mode' is disabled. It is a brand new SD card, so it is not a hardware issue. I also checked and confirmed that the lock/unlock slider on the SD card adapter is in the 'unlock' position.
Any ideas?

ryanhoetger, 2024-09-29:
I was able to edit the config by connecting via SSH and changing the connection strategy to 1 (Phone mode). My Tiguan takes a minute or two to connect, but will and is working great.

ganeshenoy, 2024-09-29:
>My Tiguan takes a minute or two to connect, but will and is working great.

This is good to know. The default value of the connection strategy variable in the latest release is 1. Were you using an earlier release?