nisargjhaveri/WirelessAndroidAutoDongle #184: RPiZW - 1+11R - XUV300

avk009, 2024-09-12:
Hey Nisarg, highly appreciate and acknowledge your amazing work!
I am on a XUV 300 W8O having the stock infotainment system and using wired android auto. I have built the wireless adaptor using Raspberry Pi 0 board with the 0.4.1 img but it doesn't connect. The 1+11r can connect to the board after booting up through BT and WiFi but disconnects the BT after few secs and keeps attempting to connect to the board with BT but doesn't succeed and the mobile goes into "looking for android auto' loop. The Android Auto does not pop up on the car's system during the whole process.
The mobile is able to connect to the board through the WiFi but is not able to SSH into it. I had to connect my laptop to the board using WiFi to get the logs. Please see the log file. ---> [ConnectAAWirelessDongle.txt](https://github.com/user-attachments/files/16978141/ConnectAAWirelessDongle.txt)

Next, I tried connecting the board to my laptop instead of the car and was able to SSH into the board to get the logs. PFA the screenshot of the logs from the mobile. ---> [image] 
The BT and Looking for Android Auto behavior remained the same in that case as well.

With this behavior I thought that the board is not able to get enough power from the car's system through the wire and so I connected another wire to the PWR port on the board but the issue remains the same and my mobile does not connect to the board to achieve a wireless Android Auto.

Hence, can you please look into the logs and help me with this issue?

Points to be noted here are:
1. Both the log files show Jan 1 as the date. So there be might be a date and time conflict. As the log file doesn't show the year, I am not able to set the date on my phone to rule this out.
2. The mobile has Type C connection and the board has a mini-USB port. Swapping between mobile and board during initial setup causes cable change. I am not sure if this might be the cause of the issue. Although, I can get a Type C to mini-USB connector to rule this out if you would suggest.

Thanks in advance, 
AVK

SurajYOLO, 2024-09-12:
Try buying a high-speed micri usb cable, i have bought Duracell usb to micro usb cable. It works absolutely fine.

SurajYOLO, 2024-09-12:
Point no2 is not a problem, connect the phone with a separate type c cable and then replace the cable which is connected with the Raspberry Pi. Within 30 secs max you'll be connected with Android auto

Niteshgarg30, 2024-09-14:
I am also using this setup on my XUV300 w8 with stock head unit. I m facing a different problem. For me the the wireless AA connects to the head unit but disconnects very frequently, and connect again. And this goes on and on. 
I have changed the cable multiple times and also using a class 10 micro sd card. 
Haven't found any solution yet.

muralirc86, 2024-10-04:
Hi Nisarg! Fantastic work.
I am trying to use this on xuv300 W8(o). Exact same issue as @Niteshgarg30. Please advise.

Mantlet, 2024-10-31:
I am facing same issue looking for Android auto 
Pi zero 2w