nisargjhaveri/WirelessAndroidAutoDongle #185: Jittery Music and Slow response on RPI 2w

hiteshgitz, 2024-09-14:
Hi,

I am currently using Raspberry Pi 2W for this project and I was able to connect to Creta system. However, the music is very jittery (skips 2/3 seconds randomly) and also the UI is very slow in response. Do you know if there is a fix already for this?

Thanks for your code.

Thanks,
Hitesh

ankitgoenka, 2024-09-14:
I am using raspberry pi 2 w with a class 10 micro sd card on hyundai creta 2020 - used it today for the 1st time for 3hrs on a highway drive - with music playing via spotify and using Google maps - made and recieved few calls too.. worked fine.. till now

nisargjhaveri, 2024-09-15:
Out best guess on the lag is wifi interference, or in some cases insufficient power from headunit so far as I've not seen any patterns of Headunit, RPi board or phone combination that always works badly. If you can, you can use a board with 5GHz wifi (Recommended RPi 4 for ease of setup) and see if that improves (no guarantees so far).

brijkishores, 2024-09-16:
I am using RPI 2w with a heat sink for my Nexon. I had frequent disconnections initially, so I figured it was because of the cable. I changed to a much better short cable, which started working better. All functionalities work fine when connected, even the initial connection happens fast when I enter the car. The issue is sometimes there is some lag when playing music while using the Maps. When I receive calls I get disconnected from the head unit. For an hour it worked perfectly with some time lags in music playing but after an hour it kept disconnecting frequently from the Head unit. It was not a heating issue as I kept the top cover open and when I touched the heat sink it was just warm. I think there is some improvement required for the RPI 2w code. And I also want to thank the developer for the solution. If it gets better I think it will help a lot of people. Also just to understand, is it possible to connect a USB port to the head unit and a separate power cable to power the RPI 2 to check if that helps?

ten24bytes, 2024-09-17:
@brijkishores can you share the link of a good short length micro USB cable? I'm unable to find one on Amazon. The reviews for most short cables are dicey on the part whether the cable supports data or not.

wsadzc, 2024-09-17:
Hello, respected developer,
I'm currently using both the Pi 4 and Pi Zero 2W interchangeably.
Indeed, the Pi 4 doesn't experience the shaking and aforementioned issues (except when it starts to overheat, which then causes shaking and disconnection).
The Pi Zero 2W does frequently encounter issues to the point of becoming unusable.
(For instance, frequent operations on the car stereo screen, such as using navigation, can cause music to stutter or pause, eventually leading to instability and the car stereo disconnecting.)
At this point, the phone screen displays a message indicating insufficient phone performance.
These malfunctions only occur with the Pi Zero 2W.
The Pi 4 runs well.
Regardless, thank you to @nisargjhaveri for the development, allowing us to use it smoothly.
I've been actively recommending it to my friends.
Currently, everything is working very well, except for the aforementioned issues with the Pi Zero 2W, which are a bit troublesome.
The above conversation was translated with the assistance of Claude.ai. I hope I've expressed my meaning clearly.
Thank you.

brijkishores, 2024-09-17:
@ten24bytes 

I bought it a couple of years back from Amazon and couldn't find the product now when I searched for it. I think I bought from a friend's account so I don't have it in my order history as well.

baudhayan, 2024-09-21:
I just bought a RasPi Zero 2W & powered it using the short data cables that used to be shipped with last generation Nokia phones. 

Works fine, except for random music stutters & connectivity issues. Thank you so much to @nisargjhaveri . The WiFi connetion shows "Low Quality" on the phone, not sure why. Hopeful that you will introduce more stability in the code in near future.

Thanks once again.

snaveenkumareee, 2024-10-15:
First of all, thanks to @nisargjhaveri for the amazing project

After the above discussion.. I observed the disconnection happens mostly at the signals, when other cars/bikes are close by. hope this is due to interference.. how can this be handled. once bluetooth and wifi are connected to the headunit, do the code still scans for new connections ? a doubt here...

ten24bytes, 2024-10-15:
> I observed the disconnection happens mostly at the signals, when other cars/bikes are close by. 

This is true even with my other car's wireless AA OEM head unit. It happens occasionally at a few signals when I'm stopped. But off late that has been a rare occurrence probably with the AA app update. Maybe that's the reason you're also facing disconnections.

ten24bytes, 2024-10-21:
> sorry did not get you ( @ten24bytes ). AA app means the android app on my mobile ? If yes, my app is on the latest version. If it is the Dongle build, I am using current-1 version. Is there any fix for this use case on the latest version?

AA = Android Auto. I have updated my comment for better clarity.

Mantlet, 2024-10-31:
I Also thing some improvement required in pi zero 2w code
AAwireless 2 is launched can its code is used in that