nisargjhaveri/WirelessAndroidAutoDongle #187: For Tata Nexon some bugs and updates for Pi0w

aarshaypardesi, 2024-09-14:
Work around for looking for Android auto

No need to connect to wifi of board. Only connect to wifi. Connect Bluetooth directly to the car. If needed activate bluetooth internet sharing 

Device not found issue

Find a cable that has 3a plus 480 Mbps features that works. Plus just check the car's usb connection. It may need to be snug fit there. A loose connection would be an issue 

Issue not resolved only one

The board or the OS has no standby disconnect option. Drains the power even after the car is switched off. The wifi continues to be switched on draining life from car battery. Had a hard time identifying this, identified this only after jumpstarting the car and then changing the car battery as it was totalled after being connected for 72 hours to board.

Rest all function works as they should be even the OK Google thing.... Calls.... Sms... Maps.... Ets etc...

If any specific issue or step needs details let me know. 

I only tried with Pi0w for now would try with other boards may be later ....

Happy DIY'ng people...

nisargjhaveri, 2024-09-15:
The expectation is that the car cuts off power when off for a long time. If that is not the case with your car we currently don't have a good solution apart from disconnecting from USB when leaving the car, unfortunately.

aarshaypardesi, 2024-09-15:
Yes that's what I understood so buying a connector to be connected to Pi
board and the small wire of the car so that to prevent either of the ports
get damaged due to multiple pulling out procedure.

Besides only this issue the board works perfectly with the OS installed if
we follow the steps properly

rajeshpanda, 2024-10-06:
Just a thought here. is it possible to enable gadget port only when the device is powered from the power port? Or use power only from power port? This way you could power the pi from lighter port while still keeping the data port connected.

rajeshpanda, 2024-10-06:
Did some digging around the pi zero boards. Usb power lines are hardwired to the 5v lane. So there is no way to change it via software. Using a usb breakout board might be the best solution here.