nisargjhaveri/WirelessAndroidAutoDongle #189: [Not an issue] Performance comparison on different Pi models?

ten24bytes, 2024-09-17:
First of all, I would like to extend my gratitude to @nisargjhaveri for creating this project.

I tried this on my Pi Zero W and Pi Zero 2W (Car: Honda City 2022, Phone: Galaxy S22 Ultra). However, the avg performance difference measured over 10 cold starts was just 3-5 sec (avg boot time 22 sec). Sometimes the Pi Zero W starts faster than Pi Zero 2W and vice versa. But overall, I can't say that there is any significant performance difference between the two boards even though 02W is almost 50% faster than 0W.

I am willing to purchase a Pi 4 and try it on that, but my reservations are whether I would get any reasonable performance benefits and whether the car USB port would be enough to keep the Pi 4 powered on.

Can someone throw some more light on this?

@nisargjhaveri Do you have any test data for the performance of various boards?

wsadzc, 2024-09-17:
Hi, I'm glad to answer your questions.
I'm using a Pi 4 for testing.
The version is 0.41 (Pi 4, Pi Zero 2W).
The car stereo is Pioneer 5050BT, which takes about 22 seconds (matching your test results).
However, the Pi 4 tends to overheat (I haven't installed any cooling devices).
The Pi Zero 2W freezes and disconnects when the car stereo screen is tapped multiple times.
The exact cause is still unknown, but the Pi 4 is indeed more stable.
Here's a summary:
The Pi 4 can be powered via USB and has better response and stability with the car stereo.
Compared to the Pi Zero 2W, the 2W is more prone to freezing, but there are no other differences.
The above conversation was translated with the assistance of Claude.ai. I hope I've expressed my meaning clearly.
Thank you.

ten24bytes, 2024-09-18:
> Hi, I'm glad to answer your questions. I'm using a Pi 4 for testing. The version is 0.41 (Pi 4, Pi Zero 2W). The car stereo is Pioneer 5050BT, which takes about 22 seconds (matching your test [...]

Thanks for your in-depth reply. I will try out Pi 4 in the future if I face problems with my Pi Zeroes 🙂

LorisTecnology, 2024-09-19:
Using it on Rpi4 as well on Peugeot 208 ii with Redmi note 10 Pro on Pixel OS A14
Is stable. First boot takes no longer then 22 seconds and AA starts on car unit. Way more stable then cable connection

Mantlet, 2024-10-31:
I dont think pi4 would be a good option you can get dongle in that price