nisargjhaveri/WirelessAndroidAutoDongle #190: Feedback on Pi 4 and Pi Zero 2W Performance with WirelessAndroidAutoDongle

wsadzc, 2024-09-17:
Hello, respected developer,
I'm currently using both the Pi 4 and Pi Zero 2W interchangeably.
The Pi 4 doesn't experience any shaking or disconnection issues
(except when it starts to overheat, which then causes shaking and disconnection).
The Pi Zero 2W, however, frequently encounters issues to the point of becoming unusable.
(For instance, frequent operations on the car stereo screen, such as using navigation, can cause music to stutter or pause, eventually leading to instability and the car stereo disconnecting.)
At this point, the phone screen displays a message indicating insufficient phone performance.
These malfunctions only occur with the Pi Zero 2W.
The Pi 4 runs well.
Additionally, the config disk partition of the Pi Zero 2W 0.41 img cannot be read in Windows,
so it can't be configured. It displays normally under Linux OS.
Regardless, thank you to nisargjhaveri for the development, allowing us to use it smoothly.
I've been actively recommending it to my friends.
Currently, everything is working very well, except for the aforementioned issues with the Pi Zero 2W, which are a bit troublesome.
The above conversation was translated with the assistance of Claude.ai. I hope I've expressed my meaning clearly.
Thank you.

wsadzc, 2024-09-17:
Regarding the power supply issue with the Pi Zero 2W:
Some users have reported needing to connect two cables for normal operation.
Here's a possible solution:
First, note that the Pi Zero 2W has two Micro USB ports: a PWR port and a port with OTG functionality.
The solution involves using a single USB cable, splitting it into 4 wires, and connecting them to individual positions: 5V, GND, D+, and D-.
This should resolve the issue.
I'm providing this method for reference.
I hope to get responses from others on this solution.
Thank you.
[image]

wsadzc, 2024-09-18:
Today I tested the method of soldering to the test points on the back of the Pi Zero 2W
Unable to communicate
Then some users also reported that they needed to plug in two wires to have communication
The possible reason for this part is that when connecting to the PWR MicroUSB
Only then will the Zero correctly execute and open MTP mode
While connecting to the OTG MicroUSB, for unknown reasons, it cannot execute to MTP mode
I used two Zero 2Ws and SD cards for cross-checking and found this unstable situation
P.S. This problem is judged to possibly be due to using some charging cables that short D+ and D- together, the cable needs to be replaced
Additionally, the audio intermittent problem is obvious, ruling out poor 2.4G signal issues
(During testing, the Pi Zero 2W was placed in an air conditioning vent, and the phone was placed close by, about 10cm away)
Ruling out overheating and interference problems
But it still occurred multiple times when operating the car stereo panel or playing music on the phone,
Resulting in audio gaps, or direct interruptions, or Pi restarting

corsicanu, 2024-09-19:
Can you please share what car do you have? What values are you getting from the car (V/Ah)? 

I'll try this myself later today, i own a Skoda Octavia from 2023, which uses type-C cable so it will be a bit tricky to convert type-C to micro USB to get to no adapters needed. 

I didn't measure yet the values, but i noticed my Samsung s24 using fast charge (which usually means around 15-18w, most likely 9V at 1.65-2Ah), will post also the measurements and the results once i finish building the project.

wsadzc, 2024-09-19:
The car models I currently own all have (Type-A) outputs with 5V voltage, and they do not use the USB protocol to step up to 9V.
Only when both ends are Type-C can they successfully handshake and increase the voltage.
I have already purchased a MicroUSB to Type-C adapter.
I will attempt to use a cable with Type-C on both ends to drive it.
If successful, you will be able to refer to my solution.

corsicanu, 2024-09-19:
I managed to connect my pi with a single short Asus micro-usb cable and a Samsung type C OTG adapter. So far it works fine, will test it in the next days to see if it starts lagging or stuttering long term. In my 10-15 mins tests everything was fine on both desktop-head-unit and car. As for the measurements i checked when running tests on the pc and it was pulling like 5V at 0.2-0.3A.

I wanted to mention one issue i had in the past, may not be the case now, but it can become pain in the a** it left untreated - microsd card, maybe try another one or a new one - i had issues to the point where my pi zero 2W was freezing entirely after 40-50 mins, only solution to that being a reboot. Except i had it at 100km away, running headless as vpn for the ipcams i had there, so reboots were totally impossible.

Will come back in the next days to share my experience with it. 
Regards.

wsadzc, 2024-09-19:
Does it include music playback functionality?
The Pi Zero 2W works well when only running map navigation.
However, if I use the Brave browser on my phone to play YouTube,
it experiences delays or interruptions.
On the same car, using a Pi 4 2GB has fewer or almost no issues,
with only occasional interruptions for unknown reasons.
Additionally, I've just tested using a micro to Type-C adapter,
and it works.
However, the interruption problem still bothers me... I'm not sure if it's the browser playback causing transmission issues.
If anyone has answers to my concerns or confusion, I would be very grateful.
Q1. Communication problem with D+ D- on the back of Pi Zero 2W
Q2. Browser playback of YouTube on Pi Zero 2W causing delays or interruptions
Q3. Occasional interruptions and delays on Pi 4 (much less frequent compared to 2W)
As someone who develops software in C#, I'm not very familiar with embedded systems operations.
If there are better methods or ways I can help, please let me know.

corsicanu, 2024-09-19:
This is my setup and i've been playing YT Music for around 45 mins with no stutters and no major lag (it has a very small delay, but nothing out of the ordinary).

[image]
[image]

wsadzc, 2024-09-19:
OK. I've tested it. Indeed, it works well under YouTube Music (played for about 5-10 minutes). During playback, the car's display screen was normal. P.S. YouTube Music forces foreground playback, background play is not allowed.
However, when switching to the Brave browser to watch YouTube, delays and stuttering quickly appeared. The car's navigation screen also experienced delays. I can't determine where the differences lie between the two. The phone's performance should be sufficient to run it (model: OnePlus 11 with 16GB RAM/512GB storage).

wsadzc, 2024-09-27:
Raspberry Pi Zero 2W Progress Update - 2024/09/27
Q1: Pi Zero 2W requiring both PWR and OTG cables for stable operation
A1: This issue is caused by a damaged MicroSD card. Replacing the SD card resolves the problem.
Q2: Communication issues with D+ and D- pins on the back of Pi Zero 2W
A2: As above, replacing the SD card resolves this issue. After replacement, both D+ and D- pins, as well as the back-side soldering shown in the image, function as expected.
Q3: Pi Zero 2W audio stuttering and frequent system freezes/lag
Solution: Implemented the Performance boost #197 solution for Pi Zero 2W.
This perfectly resolved the issue (stable operation for 20 minutes without lag).
Additional Note: Almost all dual-cable issues were found to be caused by faulty SD cards.

Paako89, 2024-09-29:
HI guys. I’m facing similar issue with a pi zero 2W. it keeps disconnecting and reconnecting to the headunit. I tried using both usb ports for more power but still no luck. the usb sign comes on for few seconds and the disappears

wsadzc, 2024-10-02:
@Paako89 
Currently, I've conducted extensive testing on the Zero 2W, but it's still unstable. I can't pinpoint where the problem lies. I'm starting to suspect that the vehicle's USB itself might be faulty... but the stability is very high on the Pi 4.
Or the SD card might be damaged, I'm not sure. As shown in the image, it's a 32GB Kingston microSD (all brand new, just unsealed, but manufactured a long time ago).
10/02 Tested the following items:

Connected the Zero 2W's Vbus and GND to an external power bank to avoid unstable power supply from the car.
(The result is still unstable, and connection is very difficult)

I'm currently waiting for the Pi 4's cooling case to arrive... the instability of the 2W has made me lose confidence.
Additionally, I tried to configure the Radxa ZERO 3W RK3566 (WiFi 6/BT5.4/eMMC).
Failed, as Windows 10 1809 LTSC couldn't install Docker for Windows.
The WSL Ubuntu 22.04 configuration also had issues, which was challenging for me.
In any case, I'll keep trying. If anyone else has completed this successfully, please make sure to share. Thank you.

Demonhawk14, 2024-10-03:
I was originally have issues with my PI zero 2w being unable to connect to my Sync3 system. Rather than plugging a usb cable into each of the ports on the PI itself, I used one of these style cables https://www.amazon.com/gp/product/B098L4HTVG/ to provide extra power on just the OTG connection. That allowed me to have it connect pretty consistently, so that might be an alternative for people experience connectivity issues due to insufficient power. Now I'm just trying to troubleshoot stuttering audio from audible. I might need to try an overclock next.

NabeelUppel, 2024-10-25:
I tried a new SD card like you suggested and it didn't seem to work. The cable I used worked before and it just kinda stopped one day.  Any other suggestions?

NabeelUppel, 2024-11-05:
Hi @wsadzc 
I have a question and it seems that you have context on the power usage/requirement

I have a Pi02w and it stopped working all of a sudden and I cannot get it to work again I see that a lot of people have issue with Pi02w and have better stability with a Pi4.

From my understanding the Pi4 is more powerful and most likely requires more power to operate so how is it more stable than the Pi02w? 

I want to make sure that I have a better shot with a Pi4 before biting the bullet and buying one. 

I did try a new SD card with no luck. 

I was testing @manio's new rust implementation and realised that my pi crashes when switching to a accessory mode and the most likely reason is power issues as far as we can tell. (Still investigating)

corsicanu, 2024-11-05:
> I have a Pi02w and it stopped working all of a sudden and I cannot get it to work again I see that a lot of people have issue with Pi02w and have better stability with a Pi4.

I had same issue and got it fixed by burning latest build from actions artifacts (https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/actions/runs/11149049881)

Seems that rootfs gets corrupted after a while with sudden power cut and commits that make rootfs read only fixed that (for me at least). Give that a shot and see long term, at first i had issues after 2 weeks of daily usage and now it's more than 3 weeks with no issue.

Regards

NabeelUppel, 2024-11-05:
Thanks @corsicanu 
I did try but with no luck, it still flashes then goes into looking for Android Auto.

Also how do you change the connect type ? I couldn't make any changes because everything was readonly.

corsicanu, 2024-11-05:
> Thanks @corsicanu I did try but with no luck, it still flashes then goes into looking for Android Auto.
> 
> Also how do you change the connect type ? I couldn't make any changes because everything was readonly.

From my experience, dd the image, boot, connect with laptop to that wifi, ssh within it as booted and do changes via cli, then reboot and profit. 
That's how i OC'ed everything a bit for better responsiveness as it's explained in #197, and that's how it might work changing type. Check in the commits for new location of configuration files as it got changed in the artifact i linked, rootfs is ro but it has includes from another place.

Regards

puregreen59, 2024-11-14:
Can I ask a stupid question, somewhat related to trying to fix my stuttering issues, I'm attempting to overclock my zero w 2 following instructions, but when I attempt to even cd to /boot/ it's unable to find the dir.

Cheers!

manio, 2024-11-14:
the boot is not available "out of the box", you can mount it eg in /mnt like this:
`mount /dev/mmcblk0p1 /mnt`
then you can go to /mnt and do `ls`

EduKap, 2024-11-16:
I have a Volvo V60 MY21 running with a Raspberry Pi Zero 2W and it works really well overall. Bootup is fast, connection initialization is consistent and the connection is reliable, never drops out. My only gripe is that Spotify music seems to skip consistently, like it's lagging. I will try with the overclock suggestions and see if it improves. I always run Waze, haven't tried Spotify without Waze yet.

manio, 2024-11-17:
I am also using Pi Zero 2W but I cannot say it is working reliable (many times it was [not starting](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/196), when I go out of range and it disconnected, then I had to restart the Pi/aawgd to get it working back again - it was not able to recover connection), that's why I [rewrote it in Rust](https://github.com/manio/aa-proxy-rs) and now I can say that it's finally working rock solid stable for me! :)

I have to say, that when it was connected using aawgd, then **mostly** it was also working fine (no cpu-like performance problems, etc), but sometimes I encounter lagging and disconnections too - but I have to admit that it was very rare.

I am using it mainly with navigation (google maps, waze) and spotify at the same time.

EduKap, 2024-11-17:
> I am also using Pi Zero 2W but I cannot say it is working reliable.

What phone are you using? I edited my previous reply with my phone (Pixel 6a) if it perhaps it related.

manio, 2024-11-17:
> What phone are you using?

moto g84 5G, before I was also using it with Samsung Galaxy A10

manio, 2024-11-17:
Regarding Pixels: https://www.androidpolice.com/google-pixel-owners-report-android-auto-connection-issues/
Isn't it your case?

EduKap, 2024-11-17:
> Regarding Pixels: https://www.androidpolice.com/google-pixel-owners-report-android-auto-connection-issues/
> Isn't it your case?

I may be in the minority but my connection has always been rock solid, both before discovering this project (running wired against my V60's head unit) and after (wirelessly via Rpi Zero 2w).

My only issue is Spotify audio skipping/lagging while using Waze. I will need to try and see if it lags when only using Spotify (not Waze), or if a small overclock can solve it.

It may also be due to the RPi Zero 2w connecting with 2.4GHz wifi - bandwidth issues?

manio, 2024-11-18:
> It may also be due to the RPi Zero 2w connecting with 2.4GHz wifi - bandwidth issues?
The bandwidth is really small comparing to what 2.4GHz wifi could handle. If you're suspecting this, I would rather bet some interferences in noisy environments or with BT...

manio, 2024-11-18:
Update:
As my aa-proxy-rs is showing the transfer, so I just collected these logs with running Waze navigating and playing Spotify at the same time. I am providing it for the reference (this is the whole 6-7 minutes trip):
```
1970-01-01, 00:01:16.217 [INFO]  proxy:  📱 phone to car transfer: 23.4 MiB (399.8 kiB/s), 23.4 MiB total
1970-01-01, 00:01:16.293 [INFO]  proxy:  📲 car to phone transfer: 65.1 kiB (1.1 kiB/s), 65.1 kiB total
1970-01-01, 00:02:16.223 [INFO]  proxy:  📱 phone to car transfer: 27.3 MiB (465.8 kiB/s), 50.7 MiB total
1970-01-01, 00:02:16.347 [INFO]  proxy:  📲 car to phone transfer: 75.1 kiB (1.2 kiB/s), 140.2 kiB total
1970-01-01, 00:03:16.238 [INFO]  proxy:  📱 phone to car transfer: 24.2 MiB (412.3 kiB/s), 74.9 MiB total
1970-01-01, 00:03:16.349 [INFO]  proxy:  📲 car to phone transfer: 74.6 kiB (1.2 kiB/s), 214.8 kiB total
1970-01-01, 00:04:16.240 [INFO]  proxy:  📱 phone to car transfer: 21.2 MiB (362.5 kiB/s), 96.1 MiB total
1970-01-01, 00:04:16.548 [INFO]  proxy:  📲 car to phone transfer: 74.8 kiB (1.2 kiB/s), 289.6 kiB total
1970-01-01, 00:05:16.472 [INFO]  proxy:  📱 phone to car transfer: 13.9 MiB (236.8 kiB/s), 110.1 MiB total
1970-01-01, 00:05:16.910 [INFO]  proxy:  📲 car to phone transfer: 69.9 kiB (1.2 kiB/s), 359.5 kiB total
1970-01-01, 00:06:17.155 [INFO]  proxy:  📲 car to phone transfer: 10.3 kiB (174 B/s), 369.8 kiB total
1970-01-01, 00:06:17.160 [INFO]  proxy:  📱 phone to car transfer: 91.2 kiB (1.5 kiB/s), 110.1 MiB total
```

puregreen59, 2024-11-19:
> I am also using Pi Zero 2W but I cannot say it is working reliable (many times it was [not starting](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/196), when I go out of range and [...]

Dude! 😎

 Not overly familiar with building from code and spent a good 35 minutes trying to figure out how to build 😅 .. didn't go over well to be honest, I was mearly clutching at straws.

however I found you uploaded a built binary on your git for others to try.

Early indications show It's SUPER stable no issues, Seems to work perfectly! 

so very happy no stuttering on audio, even seems to play nicer with flac playing Via plexamp

With the original aggwd I was also prone to disconnects, so far 20 mins with no disconnects

Will report back by the end of the day.

I do delivers and ride share so will get some good thorough testing before the day is over.

samirkape, 2024-11-19:
Hello,
If don't mind, can you please share the steps you performed to use rust
based aawgd

manio, 2024-11-19:
> If you don't mind, can you please share the steps you performed to use rust based aawgd

Details are here: https://github.com/manio/aa-proxy-rs/issues/1

TL;DR
Recent ready-to-use full SD image is here: https://skyboo.net/temp/aa/raspberrypizero2w-sdcard4.img
Binary only (can be build with docker): https://skyboo.net/temp/aa/aa-proxy-rs3

I think I should more intensive thing about some building infrastructure or some drop-in script...

puregreen59, 2024-11-19:
Oh are they pi zero 2W only?? I think I'm using AA-proxy-rs3 currently on my pi 4 🤣

Seems to be working ok?..

manio, 2024-11-19:
Oh... if the Pi4 image is also 32-bit ARM than it definitely can work without any issues :)

puregreen59, 2024-11-19:
> Oh... if the Pi4 image is also 32-bit ARM than it definitely can work without any issues :)

It must be? Seems to working swimmingly 😅 just added your reconnection code to my startup script and that's also working well!

Now to take it into the "real world" and not just sitting on my drive 😎

samirkape, 2024-11-19:
> > If you don't mind, can you please share the steps you performed to use rust based aawgd
> 
> Details are here: [manio/aa-proxy-rs#1](https://github.com/manio/aa-proxy-rs/issues/1)
> 
> TL;DR Recent ready-to-use full SD image is here: https://skyboo.net/temp/aa/raspberrypizero2w-sdcard4.img Binary only (can be build with docker): https://skyboo.net/temp/aa/aa-proxy-rs3 Those are for Pi Zero 2W only. If you want to test on other Pi let me know. Quick-n-dirty install guide by @NabeelUppel: [here](https://github.com/NabeelUppel/aa-proxy-rs/blob/2823a32c087018bc41d15daca08523fac3cbfaf4/README.md#steps-to-follow-to-auto-start-script)
> 
> I think I should more intensive think about some building infrastructure or some drop-in script...

Thank you @manio
Will try and update you with the results

samirkape, 2024-11-19:
Tested for sometime on elite i20 with rpi 2w.. works flawlessly! AA headunit projection is also very clear and responsive. Thanks @manio @nisargjhaveri

puregreen59, 2024-11-19:
agreed, been out for a hour with no disconnections! the only Stutter i get with my music so far is Just after connection, lasts maximum of a few seconds then its stutter free unlike the original aawgd

im attempting to set the sd card as read only and not sure if im understanding correctly where it should be placed inside of cmdline.txt, im placing it like this

"root=/dev/mmcblk0p2 rootwait console=tty1 ro"

with some googling i should be able to see if something is ro or rw using the mount command but everything says rw am i missing something

have to applause, im learning as i go .. hah..

manio, 2024-11-19:
> have to apologise, im learning as i go .. hah..

no problem this is how I did it:
https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#hardening--making-system-read-only

puregreen59, 2024-11-19:
> > have to apologise, im learning as i go .. hah..
> 
> no problem this is how I did it: https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#hardening--making-system-read-only

Which command allows me to see read only or read write like shown in your screenshots on your linked page?

So far it's been a good 6 hours in the car with no disconnects or issues at all using @manio  aa-proxy-rs3

manio, 2024-11-19:
> Which command allows me to see read only or read write like shown in your screenshots on your linked page?

It is not screenshot but diffs - the red lines are the old values, green are the new ones; so e.g. you change `rw` to `ro` in `/etc/fstab` for `/dev/root`.
After reboot type mount and you should see the first line like this:
```
# mount
/dev/root on / type ext4 (ro,relatime)
```

manio, 2025-07-28:
Good catch! :)

kadimgh, 2025-07-30:
@rickgitdone  @manio isn't WiFi managed by ```hostapd``` rather than ```wpa_supplicant``` since the dongle is in AP mode?

manio, 2025-07-30:
Oh boy... stupid me! You're damn right @kadimgh!
It is installed but commented out, so it is not even probably starting and running because hostapd is managing encryption internally!
It is there probably only for testing when connecting as a client.