nisargjhaveri/WirelessAndroidAutoDongle #192: Not working on Hyundai Venue 2019 model SX 1.0 turbo

satpalravi, 2024-09-20:
I have tried everything based on stepw, suggestions and google. 
Changed the cable, used a new card. Using PIZero2w

Problems:
1. When I plug the PIZero2W the green led light up immediately, however keeps blinking random 
2. I had to plug another cable to power to make the led light stable 
3. I get the bluetooth and wifi both visible and wifi is connected through wifi
4. Nothing comes up on the screen after all connections 
5. I have Android 14 phone, techo pova 6 pro 5g

Does it not work with Hyundai Venue ? I checked the cable is not loose phone is connecting fine.

I want to try this, hope you help fix this.!!

wsadzc, 2024-09-20:
First,

Please check if the MicroUSB is capable of data transfer
(Many MicroUSB cables only connect 5V and GND, providing charging function only)
Insert it into the second OTG port on the Raspberry Pi, where the back is labeled with DP DM and corresponding metal dot positions
After confirming that all the above functions are correctly connected, verify if the functionality is normal. This usually resolves all basic issues.

satpalravi, 2024-09-21:
yes. 
I am using a data transfer cable with 480mbps speed. I am using his cable to convert my RPi02w for pi-hole, works perfectly fine. 

The cable is inserted into the second USB port (not the power port), however, since the light was not stable, I gave power by inserting another able to the power usb slot

Bluetooth and wifi are connected to phone and stable (wifi is continually connected, does not disconnect)

even after trying everything, android auto does not appear on screen.

PS: changed the card to latest 32gb card as well.

nisargjhaveri, 2024-09-24:
#88 might be related. We still need to find a solution for the insufficient power issue on some cars which is likely causing the two usb required. Some people have reported that reconnecting usb cable while PWR is still connected helps. Obviously, not ideal.

wsadzc, 2024-09-25:
Hello @nisargjhaveri 
My colleague has described an issue with the Pi Zero 2W regarding its USB configuration. The problem manifests as follows:
1. When using a 64-bit operating system, the USB connection would abnormally disconnect and repeatedly malfunction.
2. The issue was resolved by switching to a 32-bit operating system.
However, both the root cause of the problem and the reason why the 32-bit OS solved it remain unknown.

nisargjhaveri, 2024-09-25:
@wsadzc I believe the default build we have is 32 bit only. #129 also mentions some issues with usb on the 64bit builds.