nisargjhaveri/WirelessAndroidAutoDongle #197: Preformance boost

japuzzo, 2024-09-26:
Given a new Raspberry Pi Zero 2W
Edited /boot/config.txt

`# Overclocking Rpi Z-2W slightly. Does not need heat-sink to support these slight improvements 
arm_freq=1085
gpu_freq=530
over_voltage=2
core_freq=515
sdram_freq=533
over_voltage_sdram=1`

It did improve touch response times.
Did not fix audio stuttering of FLAC files, but did help with MP3.

WARNING! 
If you do not know what over-clocking is do not attempt, do not modify!
If you do not know what you are doing, you will brick your Rasbery Pi, destroy your car's internals leading to your tragic death!
Thus DO NOT ATTEMPT anything if you don't understand what you are doing!!! 
You have been WARNED!!!

japuzzo, 2024-09-26:
Did some more tweaking after looking at: [https://www.raspberrypi.com/documentation/computers/config_txt.html](https://www.raspberrypi.com/documentation/computers/config_txt.html)

Upped the min frequency by the same about as the max. Did not force turbo because it may overheat unit.
Removed GPU setting as that is not needed. 

```
# Overclocking Rpi Zero 2W slightly. Does not need heat-sink to support these slight improvements 
arm_freq=1085
arm_freq_min=685
core_freq=515
core_freq_min=365
over_voltage=2
sdram_freq=533
over_voltage_sdram=1
```

WARNING!
If you do not know what you are doing, you will brick your Pi, destroy your car which will cause your tragic death!

wsadzc, 2024-09-27:
Here's the English translation:
I'm very happy that this has effectively improved my unstable situation.
I've also added passive cooling, otherwise the heat would be quite severe and could lead to damage.

japuzzo, 2024-10-15:
When out and purchased a heatsink for my Rpi-Z-W to test the maximum overclocking setting on the dongle:
```
# Overclocking Rpi Z 2W == with a heavy duty heatsink ONLY
arm_freq=1150
arm_freq_min=750
core_freq=600
core_freq_min=450
over_voltage=6
sdram_freq=600
over_voltage_sdram=4
sdram_schmoo=0x02000020
```
These settings will void your warranty ( the over_voltage=6 blows a fuse that's how they can tell ) so don't do it unless you know what you are doing and also have really good cooling! Since the voltage is increased, there will be more heat! 
I have not found any reports of a stable Rpi-Z with settings of greater then arm_freq=1150

japuzzo, 2024-10-15:
Heatsink used in testing:

[image]

japuzzo, 2024-12-06:
Turns out the heatsink ( although required for overclocking ) is a bit of overkill. Since the Pi-Z-W is running one program ( just a few threads ) there is no need to task switch or handle other events. So the Pi-Z-W never uses the top end of the overclocking settings. But  the "min" settings did effect the performance. Thus setting a higher min should improve performance. Exactly what setting is the best I have not discovered, but the last min settings did improve performance:  
`arm_freq_min=750`
`core_freq_min=450`

manio, 2024-12-06:
@japuzzo
I am also using [LoRa](https://github.com/manio/lora-gw-rs) on my Pi, so I needed the serial port (`dtoverlay=pi3-miniuart-bt`). And to use the serial port properly [without any problems](https://github.com/raspberrypi/linux/issues/6462) I also had to add:
`core_freq=250` so I believe it is set to a fixed value.

I am also having a similar heatsink but I think it is better to have it then not. A car can run in variety of temperatures (hot in summer) so I think it is good to have some margin.

btw: I think it is also worth considering add some extra optimizations like:
```
# disable HDMI (saves power)
max_framebuffers=0
# don't try to read HDMI eeprom
hdmi_blanking=2
hdmi_ignore_edid=0xa5000080
hdmi_ignore_cec_init=1
hdmi_ignore_cec=1
hdmi_ignore_hotplug=1
disable_fw_kms_setup=1
# disable composite video output
enable_tvout=0

# all these options cause a wait for an I2C bus response
force_eeprom_read=0
disable_poe_fan=1
ignore_lcd=1
disable_touchscreen=1
camera_auto_detect=0
display_auto_detect=0

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
#gpu_mem_256=100
#gpu_mem_512=100
#gpu_mem_1024=100

# start_cd.elf is used in the case we specify 16MB
# https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/ad334e8fe0e320baaafccdd0b9a062c43560a1b8
[pi0]
gpu_mem_256=16
gpu_mem_512=16
[all]
gpu_mem_512=64
gpu_mem_1024=64
```

Worth reading: https://kittenlabs.de/blog/2024/09/01/extreme-pi-boot-optimization/

Regarding this project: a good optimization job was also done by @joka90:
https://github.com/joka90/WirelessAndroidAutoDongle
He also optimized the kernel and went down to 5 seconds bootup :)

I have something about 7secs currently without doing kernel cut-down.

joka90, 2024-12-30:
Sorry for late reply; From what I recall configuring the app build made the biggest impact. The c++ binary was ~6.5 MB with debug info from top of mind. 2nd best was swiching away from ext4 to a filesystem which would fit in the [VFS cache](https://www.usenix.org/legacy/publications/library/proceedings/usenix01/full_papers/kroeger/kroeger_html/node8.html), I.E f2fs. Removing all modules not needed made it easier to fit (as the image was just 50 MB something). 

Then I could shave some more time by adjusting the pi boot-loader, by geting the stripped binaries, as the bootloader is slow to load the actual binaries. At this level it is not nececary to optimize the boot more as one need to wait on the car headunit to boot etc.

Regarding the lag after app -O2 and removing debuginfo/symbols: Still had issues with lag tidal + fast map move's, could see that I almost saturated the 2.4Ghz wifi, looked on diffrent ways of making the "pointer/touch" input have prio in the tcp send/recive queues of the wifi drivers and trying to figure out if it was possible to tune the tcp channels or WIFI for faster/better scaling. Used the x86 build of the [non oem mazda AA binary](https://github.com/gartnera/headunit) (ubuntu folder), to iterate faster (not needing to sit in the car).

Also tried to apply the last parts of the RT patch set to see if that could remove the last part of jitter in the wifi driver, tcp and IO stack. Started to investigate ebpf and implement the whole usb <-> tcp/ip <-> wifi part. Then got a firmware for the [AA wireless](https://www.aawireless.io/en) adapter that solved the connect issue I had (aa wireless used 10.x.x.x which was also used for voLTE when doing dual-sim, 5g data was over 2nd sim cgnat (100.x.x.x)).

Will have a look on the rust ebpf parts, but think that 2.4ghz wifi is the issue for me :(.

manio, 2024-12-30:
> Still had issues with lag tidal + fast map move's, could see that I almost saturated the 2.4Ghz wifi, looked on diffrent ways of making the "pointer/touch" input have prio in the tcp send/recive [...]

[Some](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/190#issuecomment-2485890089) [users](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/190#issuecomment-2486622364) reported, that it was better with aa-proxy-rs, you may also try it if you like...

nisargjhaveri, 2024-12-30:
Thanks for sharing @joka90. I'd love if we can get some of these in. Binary optimisation is a no-brainer, obviously, if it is not being already. :/

@manio, curious, any reason rust implementation would be better on scaling transfer speeds?

manio, 2024-12-30:
> curious, any reason rust implementation would be better on scaling transfer speeds?

In terms of speed the C is rather comparable to Rust, I am not quite sure in this case, but if I had to bet I would think of:
1. using io-uring which is fast low level kernel API for data transfer
2. not analyzing any headers/data like in [your](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/package/aawg/src/proxyHandler.cpp#L51) implementation but just simply bidirectional forwarding

But this is just a guest.