nisargjhaveri/WirelessAndroidAutoDongle #198: Configuration blues

ganeshenoy, 2024-09-28:
Disclaimer: I realise that this is not exactly an "issue" in the strictest sense of the word, but am raising it because I've exhausted all other avenues.

I'm facing this weird and irritating problem in my car where the head unit throws a 'USB: Device is not supported.' warning every time before the connection happens successfully. I was asked to try the different values available for the AAWG_CONNECTION_STRATEGY variable to see if it will help. 

Being a newbie to embedded systems, I thought it would be a simple case of plugging the SD card into my Windows laptop and updating the config file. It turns out that the file in which the AAWG_CONNECTION_STRATEGY variable is declared is located in the rootfs partition which is not visible in Windows. 

Windows Subsystem for Linux (wsl) was my next port of call. I installed it and tried to mount the SD card only to find out that wsl does not support SD cards. 

I then tried mounting the SD card using the Linux File Systems for Windows program by Paragon, and the rootfs partition is marked read-only. The 'Mount in Read/Write mode' is disabled. It is a brand new SD card, so it is not a hardware issue. I also checked and confirmed that the lock/unlock slider on the SD card adapter is in the 'unlock' position. I've hit a dead end.

I'm raising this issue in the hope that I will be provided a solution to a problem that I'm sure other users are facing too, viz. a user-friendly and hassle-free to configure the program, especially for non tech-savvy folks.

nisargjhaveri, 2024-09-28:
This is most definitely an issue. Editing configuration should be easy.

1f9253f62c5d563c2cf9eedf83f9dd9e416d926f already attempts to fix this. It moves the config file to the vfat partition which should be recognized by most OS out of the box. So, after this change, we can simply plug in the sd card and the config file should show up. You can download and try out the latest build from https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/actions/runs/11000965216

In the older versions, editing config file is an issue. If you can't mount the rootfs on sd card easily, you can ssh into the board when running and edit the file. This is also not very friendly. I hope the new updates on main branch will help.

ganeshenoy, 2024-09-28:
> [1f9253f](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/commit/1f9253f62c5d563c2cf9eedf83f9dd9e416d926f) already attempts to fix this.

This helps a lot, Nisarg. I will try it out and let you know asap.

nisargjhaveri, 2025-03-28:
Should be fixed in v0.5.0. It now exposes `aawgd.conf` directly in the boot partition that can be edited directly on the machine that prepares the SD card.

closed 2025-03-28 by nisargjhaveri