nisargjhaveri/WirelessAndroidAutoDongle #199: Apple CarPlay Request

staruthaya, 2024-09-30:
Hello @nisargjhaveri 

Thank you very much for this amazing project. Using it for a while without any issues.

Looking for a Apple CarPlay, please let me know if this is in our roadmap. Or refer any other projects that you know.

Thanks.
Uthaya

utsavsinghal, 2024-10-01:
+1, Please add support for Apple CarPlay.

gamelaster, 2024-10-20:
Making CarPlay support is problematic. The documentation are only leaked source codes from Apple, and you need MFi chip on the hardware to have functional CarPlay. Minimally, this would require additional hardware with MFi for Raspberry Pi. I think this is beyond the scope of this project. Also, getting those MFi chips legally is questionable.