nisargjhaveri/WirelessAndroidAutoDongle #201: Slow to connect

maxamilus, 2024-10-01:
I've noticed that around 20 or so seconds, my phone will recognize that it's trying to connect to android auto, but it takes around 1:20 - 1:40 min for the head unit to actually connect.  It runs fine after this connection is made.  Has anyone else had a similar experience?

I'm using the ZeroW in a Ford Bronco Sport that is running Sync 3.  Pixel 9 Pro.

nisargjhaveri, 2024-10-04:
Connection time depends on multiple things. The time BT and wifi connection takes and also on when the car becomes ready to connect via USB. If any of these fails for doesn't connect on time, it retries after some time.

If you can get the logs out, it might help understanding what is the case for you. From your description, BT seems to be connected at ~20 seconds. Wifi and USB needs to be seen.

maxamilus, 2024-10-05:
Hi, here is the log.

Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:38 UTC)
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1490-423) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:33:F7:0B vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000940] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001435] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001513] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060289] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060395] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060583] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060794] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060864] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062256] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064141] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064222] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064540] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065794] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076598] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076991] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... cut ...]
Jan  1 00:02:23 buildroot authpriv.info dropbear[229]: Child connection from 10.0.0.11:56854
Jan  1 00:02:26 buildroot authpriv.notice dropbear[229]: Password auth succeeded for 'root' from 10.0.0.11:56854
Jan  1 00:02:26 buildroot authpriv.info dropbear[229]: Exit (root) from <10.0.0.11:56854>: Exited normally
Jan  1 00:02:50 buildroot authpriv.info dropbear[231]: Child connection from 10.0.0.11:56861
Jan  1 00:02:50 buildroot authpriv.notice dropbear[231]: Password auth succeeded for 'root' from 10.0.0.11:56861

nisargjhaveri, 2024-10-06:
> Jan 1 00:00:52 buildroot user.info aawgd[216]: USB Manager: Timeout waiting for accessory start request

As expected, this suggests that the car isn't recognising and starting the accessory on usb. The second try succeeds.

My guess would be that the headunit is not yet ready to connect by then. You can try the "dongle mode" which waits for the car's bluetooth to connect with the phone., which would likely mean that the headunit is ready to connect as well.

maxamilus, 2024-10-06:
Can this be done if I SSH into the RPi, or would I need to rebuild the image?

nisargjhaveri, 2024-10-06:
Yes, you should not need rebuild. SSH and edit /etc/aawgd.env file.

If you're using a later version from the main branch, you edit aawgd.conf.sh file directly in the SD card's boot partition.

maxamilus, 2024-10-06:
It doesn't seem to connect at all in mode.  Here's the log for this.

Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:38 UTC)
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1490-423) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:33:F7:0B vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000006] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000159] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000940] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001438] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001517] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060290] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060394] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060582] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060791] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060860] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062250] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064131] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064212] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064534] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065782] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076581] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[... cut ...]
Jan  1 00:01:35 buildroot user.info aawgd[216]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_C0_1C_6A_A8_E1_C9
Jan  1 00:01:35 buildroot user.info aawgd[216]: Bluetooth device already connected, disconnecting
Jan  1 00:01:57 buildroot user.info aawgd[216]: Found 1 bluetooth devices
Jan  1 00:01:57 buildroot user.info aawgd[216]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_C0_1C_6A_A8_E1_C9
Jan  1 00:01:57 buildroot user.info aawgd[216]: Bluetooth device already connected, disconnecting

nisargjhaveri, 2024-10-06:
Make sure the phone also is paired with the headunit's BT and is connected when the headunit starts.

You may have to clean Android Auto's data on the phone and maybe remove and re-establish pairing with dongle as well.

maxamilus, 2024-10-08:
Unfortunately none of these suggestions seems to have helped.

manio, 2025-08-09:
@maxamilus you still interested with this?

maxamilus, 2025-08-10:
The issue is unsolved, however I've opted to use a different device in the mean time.