nisargjhaveri/WirelessAndroidAutoDongle #203: Unable to stay connected to BT

ChrisSewell, 2024-10-03:
Running on a Raspberry Pi 4 using a pre-compiled image.

I've tried using my phone to pair to the Pi, and tried using bluetoothctl to pair / trust to the phone. 

[bluetooth]# devices
Device 08:8B:C8:2F:XX:XX Pixel 9 Pro Fold

[bluetooth]# connect 08:8B:C8:2F:XX:XX
Attempting to connect to 08:8B:C8:2F:XX:XX
[CHG] Device 08:8B:C8:2F:XX:XX Connected: yes
[CHG] Device 08:8B:C8:2F:XX:XX ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device 08:8B:C8:2F:XX:XX ServicesResolved: no
[CHG] Device 08:8B:C8:2F:XX:XX Connected: no

[bluetooth]# info 08:8B:C8:2F:XX:XX
Device 08:8B:C8:2F:XX:XX (public)
	Name: Pixel 9 Pro Fold
	Alias: Pixel 9 Pro Fold
	Class: 0x005a420c (5915148)
	Icon: phone
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: PANU                      (00001115-0000-1000-8000-00805f9b34fb)
	UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (a82efa21-ae5c-3dde-9bbc-f16da7b16c5a)
	Modalias: bluetooth:v00E0p5013d0000

Error log:
Jan  1 00:14:06 buildroot user.info aawgd[273]: Found 1 bluetooth devices
Jan  1 00:14:06 buildroot user.info aawgd[273]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:14:06 buildroot user.info aawgd[273]: Bluetooth device already connected, disconnecting
Jan  1 00:14:11 buildroot daemon.err bluetoothd[174]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:14:11 buildroot user.info aawgd[273]: Failed to connect device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:14:11 buildroot user.info aawgd[273]: Failed to connect to any known bluetooth device
Jan  1 00:14:31 buildroot user.info aawgd[273]: Found 1 bluetooth devices
Jan  1 00:14:31 buildroot user.info aawgd[273]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:14:31 buildroot user.info aawgd[273]: Bluetooth device already connected, disconnecting
Jan  1 00:14:36 buildroot daemon.err bluetoothd[174]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:14:36 buildroot user.info aawgd[273]: Failed to connect device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:14:36 buildroot user.info aawgd[273]: Failed to connect to any known bluetooth device
Jan  1 00:14:56 buildroot user.info aawgd[273]: Found 1 bluetooth devices
Jan  1 00:14:56 buildroot user.info aawgd[273]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:14:56 buildroot user.info aawgd[273]: Bluetooth device already connected, disconnecting
Jan  1 00:15:01 buildroot daemon.err bluetoothd[174]: src/profile.c:record_cb() Unable to get HSP HS SDP record: Host is down
Jan  1 00:15:01 buildroot user.info aawgd[273]: Failed to connect device at path: /org/bluez/hci0/dev_08_8B_C8_2F_XX_XX
Jan  1 00:15:01 buildroot user.info aawgd[273]: Failed to connect to any known bluetooth device

nisargjhaveri, 2024-12-05:
> I've tried using my phone to pair to the Pi, and tried using bluetoothctl to pair / trust to the phone.

This is generally not required, any reason you're trying to manually pair and connect? With the pre-built image, you can boot up and simply connect from phone.

ChrisSewell, 2024-12-05:
The attempt to manually connect was done to try and troubleshoot why it wasn't maintaining a connection when trying to just connect via the phone, to see what was happening when the phone was connecting and subsequently being disconnected.

nisargjhaveri, 2024-12-05:
That is normal. It connects via bluetooth to exchange wifi connection information. Once that is done, bluetooth is turned off on Pi. So disconnecting after a brief while is okay.

Once wifi information is exchanged, phone should connect via Wifi and Usb connection should start. If it is not working, issue is with wither of these steps and likely not with bluetooth connection.

closed 2024-12-09 by ChrisSewell