nisargjhaveri/WirelessAndroidAutoDongle #204: Unable to stay connected in bluetooth

muralirc86, 2024-10-07:
Hi, 

Does not stay connected for more than 2-3 seconds to the phone (Galaxy A53), hence android auto is also available only very briefly before it disconnects from the head unit (XUV300 W8(o)). This repeats.
Was able to connect using laptop and reproduce the problem. In bluetoothctl,

connect 94:52:44:42:07:52
Attempting to connect to 94:52:44:42:07:52
[CHG] Device 94:52:44:42:07:52 Connected: yes
[CHG] Device 94:52:44:42:07:52 ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device 94:52:44:42:07:52 ServicesResolved: no
[CHG] Device 94:52:44:42:07:52 Connected: no

This repeats. Searching for the bluesz error gave a plausible solution: 
https://forums.debian.net/viewtopic.php?t=155520

This suggests that one needs to install libspa-0.2-bluetooth.

1. Is it possible to install this onto the image at my end?
2. Could you please see if this indeed is the issue and provide an updated image?

By the way, I am using pi zero W. Cables verified, power verified. Using the latest image (v0.4.1)

Thanks in advance,
Murali

gamelaster, 2024-10-20:
Phone after connecting to AAWireless Bluetooth should keep connected only for short period of time, afterwards it disconnects you, so you can connect to car's Bluetooth.

nowell-morris, 2024-11-08:
So this is intentional? Does that mean that I should then pair my phone to the car's bluetooth? I would think that if you pair your phone to the car's bluetooth, then every time you turn on the car, your phone will pair to the car's bluetooth and then never pair with AAWireless and start the connection.  Am I missing something?

gamelaster, 2024-11-09:
@nowell-morris Android Auto automatically pairs with the Headunit's Bluetooth by default.

But, for automatic connection, phone and Raspberry Pi  needs to connect to each other, send packet with wifi information, and then phone connects to Raspberry wifi, which does bridge with Headunit, and Headunit then instructs phone to connect to Headunits Bluetooth.

manio, 2024-11-12:
Hi,
FWIW I described the whole process here:
https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical
Maybe you find it useful ;)

sivachandran, 2024-11-12:
@manio Thank you very much for documenting the workflow :bow: . I'm thinking to understand the implementation and optimise if there is any scope. This document would be really helpful.