nisargjhaveri/WirelessAndroidAutoDongle #205: Android Headunit and HUR Support with raspebrry pi(WirelessAndroidAutoDongle)

rahuljs, 2024-10-14:
I have Android Headunit . currently using Android auto with Headunit reloaded app(https://play.google.com/store/apps/details?id=gb.xxy.hr&hl=en_IN). but when I tried to use this program. the phone shows android auto connected with raspberry pi but Head unit doesn't recognise the connected USB. can you help me to write a code  which will allow to recognise  connected raspberry pi.

gamelaster, 2024-10-20:
use Headunit Reloaded apps and tools for wireless functionality. 

If your headunit does not have wifi (and/or bluetooth), then it might be indeed problem.
Although, debugging such situation is problematic, so hard to fix it.

rahuljs, 2024-10-21:
My head unit has wifi and Bluetooth. but it gets disconnected after some time.

gamelaster, 2024-10-21:
that means connection issues. It happens when connection is through 2.4 GHz wifi.