nisargjhaveri/WirelessAndroidAutoDongle #207: Voice Commands Not Working

Velhankar, 2024-10-18:
When using voice commands, after the command is said the feature freezes, sometimes only first word is recognised before the feature freezes. Most of the time none of the commands are recognised. The same works on wired mode.

Tttyagi123, 2024-10-20:
It's a known issue but no resolution for it yet

gamelaster, 2024-10-20:
This project is just bridge between your phone and headunit. If you have such problem, it's very likely that it's problem on phone or headunit side. Also verify on which Bluetooth device you're connected.

Also, do you use RPi Zero 2W?

Tttyagi123, 2024-10-21:
> This project is just bridge between your phone and headunit. If you have such problem, it's very likely that it's problem on phone or headunit side. Also verify on which Bluetooth device you're connected.
> 
> Also, do you use RPi Zero 2W?

Not true, the developer has recognised this issue with diff headunits and phones specially when using 2w, the upcoming build may have a fix for it.

gamelaster, 2024-10-21:
> the developer has recognised this issue

Where I can find more info about this?

Tttyagi123, 2024-10-21:
> > the developer has recognised this issue
> 
> Where I can find more info about this?

https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/185

gamelaster, 2024-10-21:
Where developer says they will fix it?

Tttyagi123, 2024-10-21:
> Where developer says they will fix it?
I also said they may and not that they will

gamelaster, 2024-10-21:
Well, the fact that developer acknowledges this issue does not mean it may or may not be fixed in upcoming builds.

Due to nature of this problem, its very very very unlikely it will be fixed in software, as this is all due interference.

Tttyagi123, 2024-10-21:
> Well, the fact that developer acknowledges this issue does not mean it may or may not be fixed in upcoming builds.
> 
> Due to nature of this problem, its very very very unlikely it will be fixed in software, as this is all due interference.

As the developer pointed out it may be poorly powered raspberry pi as some headunits provide less power through data port and it seems to be the problem with my car too, i inserted another usb to the power port and the problem has come down from frequent to occasional so it may already have a fix for it.

Billhig, 2024-11-27:
I have the same problem.  The vehicle, cable, and wireless adapter used to work fine with my previous phone.  I changed phones so I don't know if its related. The new phone connects faster and everything works except voice commands.  "Hey Google" works to put it in listen mode, same as the steering wheel button so I know the microphone is communicating with the phone.  Then it freezes just like the original poster. I don't recall if the Pi is a W or 2W.  Does the problem exhibit itself on both models?