nisargjhaveri/WirelessAndroidAutoDongle #208: Raspberry pi zero 2 w no voice command

Tttyagi123, 2024-10-18:
"Ok google", activates assistant but does not listen to any command given afterwards.

Many a times while initiating connection even with high speed micro usb and data port connected on raspberry pi, the device keeps trying to find wireless android auto, even if wifi connected manually, the bt keeps on failing to connect, the only way to connect is to unplug the cable from raspberry pi and connect it back, then it connects.

Tttyagi123, 2024-10-18:
A d once connected, the music on local music app and spotify both stutter from time to time

gamelaster, 2024-10-20:
Use another Raspberry Pi which have 5 GHz WiFi, which provides higher bandwidth and also is better suited for this. That should resolve your issues.

Tttyagi123, 2024-10-21:
> Use another Raspberry Pi which have 5 GHz WiFi, which provides higher bandwidth and also is better suited for this. That should resolve your issues.

Will wait till the developer provides a fix for this issue, i know that i can buy a 5g dongle and connect to my pi but the issue is not that frequent as of now so one can live with it till it is resolved

gamelaster, 2024-10-21:
> Will wait till the developer provides a fix for this issue

Was this somewhere addressed by @nisargjhaveri ?

Tttyagi123, 2024-11-19:
Only voice command issue remains but that's working intermittently too, thanks.