nisargjhaveri/WirelessAndroidAutoDongle #209: Add Dashcam Functionality

varishtsg, 2024-10-20:
Can we add Dashcam Functionality using the raspberry pi camera to this. I am kinda already using the zero 2w using this repo: https://github.com/notchum/dashcam as a Dashcam. I recently saw that the zero 2w can double up as a wireless aa unit as well. Would be a nice combo.

thanhphat235, 2024-11-03:
> Can we add Dashcam Functionality using the raspberry pi camera to this. I am kinda already using the zero 2w using this repo: https://github.com/notchum/dashcam as a Dashcam. I recently saw that the zero 2w can double up as a wireless aa unit as well. Would be a nice combo.

i have same idea like you. Please help me step by step can add dashcam function into the buildroot of 
WirelessAndroidAutoDongle

Thank you

TheFlyingCrocodile, 2025-04-14:
Android auto has added features for vehicle dashcam integrations I'm not sure if this is an easier solution to add a camera module to via the pi but it's also a feature I would love

https://source.android.com/docs/automotive/hmi/dashcam