nisargjhaveri/WirelessAndroidAutoDongle #212: Power on the board remain on even after car turned off

atishsonkusale, 2024-10-23:
Tata Tigor
Even after car turned off , power continuously there and dongle remain on .

sivachandran, 2024-10-23:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/211 is related to this. It has solution too.

3xtant, 2025-01-30:
Yes, this is an issue for many AA adapter users. In my case, I have an EV. The data port shuts down after a few minutes when the car is off, but the port remains on while the car is plugged in, whether charging or not.

That means my phone won't connect to WiFi at home because it stays connected to AA.

Many people, me included, have resorted to a hardware solution, either using a USB cable with a power switch, or using a Y cable and a power blocker, or the one mentioned in a related thread.

However, since this device does have access to the AA data, it would be great if it would have a configurable option to turn off WiFi while the car is off, or (less desirable) when the car is in Park. It may depend on the car manufacturer but AA does have access to some basic car info. Hopefully car on/off is part of the data available, or at least Park/Drive info.

Is that feasible @nisargjhaveri ?
BTW, thank you for this project. It is awesome.

manio, 2025-01-31:
@3xtant I am also using EV, discussion regarding this problem was here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/243

manio, 2025-01-31:
> It may depend on the car manufacturer but AA does have access to some basic car info. Hopefully car on/off is part of the data available, or at least Park/Drive info.

This is not entirely true. AA is passed through directly without doing any kind of interpretation from our side. This info is probably available in the car headunit but it is not exposed to the RaspberryPi.

3xtant, 2025-01-31:
Thanks for the quick reply, @manio . I noticed several threads start with different problems but end up going towards this issue. Finding a fix that does not require external hardware would be great.
It is a bummer that the dongle does not get any car data. I was hoping that would be the way to go.

OptimusGREEN, 2025-03-01:
Just a thought as I am having this issue too but would it be possible to provide a list of ssid that when available to the pi it doesn't try to connect to your phone?

for instance, pi sees my-super-home-wifi-ssid so doesn't try to connect to phone? 

At least maybe as an option which defaults to off? I personally wouldn't mind the AA not connecting til after I'd driven away from my home if it meant my car doesn't try to hijack my wifi connecting while I'm in my house.

manio, 2025-03-01:
> Just a thought as I am having this issue too but would it be possible to provide a list of ssid that when available to the pi it doesn't try to connect to your phone?

Oh... this is really good idea! :)
It can be done with simple bash/sh script I think... The script could search for networks in some time-period and if it find the specified then it could stop `aawgd`, then it can wait until it is not listed and start it...

For my daemon I was also thinking about checking client RSSI (signal level of the phone), but it only works for the bluetooth and bluetooth is enabled only for initiating session so this won't be reliable, but your idea seems to be a good workaround... :)

Let me know if you want some help with this script...

OptimusGREEN, 2025-03-01:
Hi, thanks for replying. Yeas I would love some help as I can't do bash lol. I wouldn't know where to start, continue or end 😂

nisargjhaveri, 2025-03-02:
Curious, if the issue is only about Pi trying to connect to phone, dongle mode should help? Does it? It should only try connect once the phone is also connected with the car's bluetooth.

manio, 2025-03-02:
> dongle mode should help? Does it?

Yes, that's true, but there are two downsides:
1. When you're connected you have to manually disconnect AndroidAuto (remembering to disable WiFi and Bluetooth before that)
2. You have to manually connect to the dongle when you want to use it (which is very inconvenient)

So trying to solve it the other/more automatic way...

OptimusGREEN, 2025-03-02:
I did actually try dongle mode a few days ago but it didn't seem to work very well for me so I switched back.

manio, 2025-03-03:
@OptimusGREEN try this: https://gist.github.com/manio/028c5ebc1e791430743b29f0a25021b5

OptimusGREEN, 2025-03-03:
Thanks @manio I'll give this a try and let you know how i get on.

OptimusGREEN, 2025-03-03:
Ok so I tried to use this but I'm a bit stuck.

I figured id do a crontab -e and set the script to run @reboot after having done a chmod +x but get this...

```
crontab: can't change directory to '/var/spool/cron/crontabs': No such file or directory
```

so i created the directory path but its gone after reboot so assuming its read only mounted into a tmp or something.

I figured I'd run the script manually to see if it killed the wifi but nothing seems to happen.

Sorry if this seems like basic stuff but not really sure where to go next.

manio, 2025-03-04:
Oh... why are you using crontab? This scripts is designed to run indefinitely. Just put in normal startup scripts (/etc/init.d/...) if you want to start it automatically. But it has to be working even when you start it manually. I tested it and it is working...

OptimusGREEN, 2025-03-04:
I'll give that a try thanks.

manio, 2025-03-04:
> I take it that directory is writable and persistant?

/etc/init.d? it is the root filesystem on the Pi. If you are doing it on running system you have to remount it as `rw`.

OptimusGREEN, 2025-03-04:
Ok, i got that to work and it indeed kills the connection to AA but not the wifi connection between phone and pi.

I'm wondering if i should change it to set wlan down but then I guess I'd have to leave my drive to ssh in for any reason lol.

I'm not sure how any of this works but think I read somewhere that the bluetooth connects first to init the wifi connection or something? would killing bt be the answer?