nisargjhaveri/WirelessAndroidAutoDongle #213: Firmware from Chinese AndriodAuto (and Carplay) Dongle

hkfuertes, 2024-10-25:
@nisargjhaveri Just letting you know I was able to download the OTA update of one of this cheap Chinese android auto wireless adapter from aliexpres onto my machine. I just `unsquashed` it and I'm going over it looking for something interesting... but my decompilation skills aren't that good.

Here is the script (taken from the update page of the dongle) to download. You have 2 models to download (change the version and custom fields at the begining of the script):
```html
<script>
    //Version is figured out via `queryDeviceVersion` whith out any Version.

    // One ...
    // const Version = "se_202409241744"
    // const PlatForm = "v851se";
    // const Custom = "jiuchangxin";

    // Another ...
    const Version = "se_202409241804"
    const PlatForm = "v851se";
    const Custom = "luruibao";

    window.download = () => {
        url = "http://120.79.59.57:8080/device-web/upgrade/downLoad";
        var xhr = false;
        try {
            var param = {};
            param["version"] = Version;
            param["platform"] = PlatForm;
            param["custom"] = Custom;

            console.log(param)

            xhr = new XMLHttpRequest();
            xhr.responseType = "blob";
            xhr.open('POST', url, true);
            xhr.setRequestHeader('Content-Type', 'application/json');
            xhr.addEventListener("progress", function (event) {
                if (event.lengthComputable) {
                    var percentage = Math.round((event.loaded * 50) / event.total);
                    console.log(`Downloaded ${percentage}%`);
                }
            }, false);

            xhr.onreadystatechange = function () {
                if (xhr.status === 500) {
                    alert("Server internal error.");
                    window.stop();
                }
                if (xhr.readyState === 4) {
                    if (xhr.status === 200) {
                        console.log(xhr.response);
                        var blob = xhr.response;
                        if (blob) {
                            var dk = document.createElement('a');
                            dk.href = window.URL.createObjectURL(blob)
                            dk.download = Version;
                            document.body.appendChild(dk);
                            dk.click();
                        };
                    } else {
                        alert('download file fail');
                    }
                }
            };

            xhr.send(JSON.stringify(param));
        } catch (failed) {
            xhr = false;
            //alert("Server internal error.");
        }
    }

    window.queryDeviceVersion = () => {
        try {
            request = new XMLHttpRequest();
        } catch (failed) {
            request = false;
        }

        if (!request)
            alert("Error initializing XMLHttpRequest!");

        var url = "http://120.79.59.57:8080/device-web/upgrade/queryDeviceVersion";
        request.open("POST", url, true);
        request.setRequestHeader('Content-Type', 'application/json');
        request.onreadystatechange = function () {
            if (request.readyState == 4) {
                if (request.status == 200) {
                    var response = request.responseText;
                    var myJSON = JSON.parse(response);

                    console.log(response);
                }
            }
        }

        var param = {};
        //param["version"] = Version;
        param["platform"] = PlatForm;
        param["custom"] = Custom;
        request.send(JSON.stringify(param));
    }

</script>
```

hkfuertes, 2024-10-25:
Save it onto an `html`, open it on a browser and just use `download();` and `queryDeviceVersion` from the developer tools.

hkfuertes, 2024-10-25:
... The magic might be happening in `/usr/bin/sdsdk` and `/usr/bin/sdDongle` which seems to be the CarPlay part of the dongle. In the rc.init I don't see any android auto "starter script" only the carplay (actually, sdsdk when loading bluetooth), but somehow the dongle knows that the connected device is an android device and changes to android auto mode...

I don't know how to continue... maybe is usefull for someone :)

hkfuertes, 2024-10-25:
I have 3 dongles. 2 of them are equal, the script above retrieves the firmware, but the third is different, with a different folder structure. You can get it here:
https://cpbox-abroad.oss-us-west-1.aliyuncs.com/2921/version.json
https://cpbox-abroad.oss-us-west-1.aliyuncs.com/2921/update.img

In this case is just only the app, without any usb gadget config, but again I don't know what to do with gHidra to make it useful...

USAvery, 2025-02-26:
> I have 3 dongles. 2 of them are equal, the script above retrieves the firmware, but the third is different, with a different folder structure. You can get it here: [...]

How were you able to unsquash the img file? I have tried with no success.

un4saken, 2025-03-14:
another device if anyone is interested

`  const PlatForm = "v851s";
  const Custom = "suding";`

un4saken, 2025-03-14:
Another device in anyone is interested
```
 const PlatForm = "v851s";
 const Custom = "suding";
```

manio, 2025-04-01:
Guys, is the firmware in those devices able to enable "Developer mode" like in AAWireless?
I am asking because we're trying [to change name](https://github.com/manio/aa-proxy-rs/issues/21) to enable this feature.

jwagnervaz, 2025-09-25:
I have ported a new version for my device  that have been abandoned  after sometime by then.

Its on my repository 

Its carplay + android play  implementation but. Car play requires a chip present in apple devices for auth.
But is easy to solder on the board.

Android Car soon will demand a cert file to be connected to newer android auto versions. 
So I will port it when needed 😉

Hope it helps someone!

gxcreator, 2025-10-25:
I have another one from Aliexpress. Managed to get adb shell and then uploaded telnet binary

```
Linux CarPlayBox 4.9.191 #1203 PREEMPT Thu Jul 31 03:53:52 UTC 2025 armv7l GNU/Linux

DISTRIB_ID='tina.masnuo.20250728.121733'
DISTRIB_RELEASE='Neptune'
DISTRIB_REVISION='5C1C9C53'
DISTRIB_TARGET='v851s-perf2/generic v2.1'
DISTRIB_DESCRIPTION='tina.masnuo.20250728.121733 4.0.0'
DISTRIB_TAINTS='no-all mklibs busybox'

./adb shell cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 48.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : sun8iw21
Revision        : 0000
Serial          : 0000000000000000
```

gxcreator, 2025-10-26:
Reading flash using xfel in FEL mode:

```
xfel.exe spinor read 0x00000000 16777216 spinor_dump.bin
```

Partition map:
```
0x000000000000-0x000000060000 : "uboot"
0x000000060000-0x0000002a0000 : "boot"
0x0000002a0000-0x000000730000 : "rootfs"
0x000000730000-0x0000007b0000 : "rootfs_data"
0x0000007b0000-0x0000007d0000 : "env"
0x0000007d0000-0x0000007e0000 : "private"
0x0000007e0000-0x000000ea0000 : "app"
0x000000ea0000-0x000001000000 : "UDISK"
```

Mountpoints:
```
/dev/root on /rom type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=30128k,nr_inodes=7532,mode=755)
/proc on /proc type proc (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
sys on /sys type sysfs (rw,relatime)
/dev/by-name/rootfs_data on /overlay type jffs2 (rw,relatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/workdir)
/dev/by-name/app on /mnt/app type squashfs (ro,relatime)
tmpfs on /run type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
/dev/mtdblock7 on /mnt/UDISK type jffs2 (rw,relatime)
tmpfs on /data type tmpfs (rw,relatime)
none on /sys/kernel/config type configfs (rw,relatime)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
```

breakgimme, 2025-10-29:
@gxcreator sure, here's the update image downloaded: https://r2.piwo.rocks/ax_202509041025

to extract squashfs: `dd if=ax_202509041025 of=root.sqfs bs=1 skip=3219520 count=11437887`

my device shipped with this exact version already so i don't have anything lower

gxcreator, 2025-10-29:
@breakgimme Seems to be Ingenic X1600E CPU + aic8800D80 WLAN combo

gator8125, 2025-10-29:
[image]
[image]

breakgimme, 2025-10-29:
@gxcreator oh nice, didn't notice that, although i think i might have tried doing something like this and had issues? might have been just my skill issue though
my setup for "reverse engineering" was me just poking around with grep and strings while sitting awkwardly with my laptop in my relatively small car though
might just go grab the device to check whether that will work

gxcreator, 2025-10-29:
@gator8125 do you want to change wifi network name or?

gxcreator, 2025-11-02:
@jamarju If you could connect using adb, you'll get boot log and partition tables. It could also be defined in DTS/DTB or somewhere inside init scripts in rootfs.
Thanks for your findings! Could you also add photo of dongle PCB and product link somewhere?

Pinball3D, 2025-12-26:
> Just got another one, seems XUDA now uses V821 RISC-V CPUs
> 
> > 
> > [image]
> > [image]
> > 
> > [image]
> 
> 
> ```
> DISTRIB_ID='OpenWrt'
> DISTRIB_RELEASE='5.0'
> DISTRIB_REVISION='r0-204c0a5'
> DISTRIB_TARGET='v821-lybox_rtl8733/generic Generic'
> DISTRIB_DESCRIPTION='OpenWrt 21.02-SNAPSHOT'
> DISTRIB_TAINTS='no-all mklibs busybox'
> DISTRIB_INFO='tina.tenc.20250625.122247'
> TINA_VERSION='5.0'
> ```
> 
> 
> 

How did you get a shell or anything on this model? I have a very similar one that and I want to mod it somehow.

breakgimme, 2025-12-26:
i see, i'm personally interested in carplay which is why i'm interested in these chinese dongles in particular

gator8125, 2026-01-12:
I still haven't managed to unpack the firmware on one of these to alter the Bluetooth name. Anyone have some guidance? I can get the PC to read it but that's it

gxcreator, 2026-01-13:
> Sorry for kinda jumping into your thread… but honestly, what’s the point of messing with stuff like this when there are better options out there? For example, that last board is based on Realtek, which works pretty poorly compared to something like the AIC8800 (at least there you can use Bluetooth and Wi-Fi at the same time, and it actually works really well). For instance, the Milkv Duo S with RISC-V is pretty cheap (around $14), and you can use it not just for AA but for a bunch of other cool projects too...

Point is to reuse very good and cheap form factor dongles. Some of them come with AIC8800, but afaik it is not better then realtek combo.

downloadme4free-gif, 2026-03-03:
Hi, sorry for any disturbance. I have a dead AA dongle (it lost power while upgrading the firmware). Is there any chance to revive it by uploading one of the firmware images from this post? How should I begin? Thank you!