nisargjhaveri/WirelessAndroidAutoDongle #216: Issue with pi zero 2w

Mantlet, 2024-10-31:
Im using pizero 2w for this 
Issue i faced
#It connected initially, but phone was getting very heatup it doesn't allow me to open apps
# later when i connected different day it doesn't get connected and getting message looking for Android auto.
[image]

salviakshay24, 2024-11-01:
Try with a diffrent micro usb cable. I had same issue and changing cable fixed it

Mantlet, 2024-11-01:
it was working with same cable before
which cable u used i have already tried 2 3 cables

zavscom, 2024-11-05:
Check the settings on the Head Unit. When the cable is connected is it in charging mode or AA mode. I have the same issue in my XUV500 and I have to changed this setting everytjme I restart the car,If the car is switched off for more than  30 min and it takes2-3 min to pick it up.

zndtoshi, 2025-01-12:
> it was working with same cable before which cable u used i have already tried 2 3 cables

Did you find a solution?