nisargjhaveri/WirelessAndroidAutoDongle #217: Android auto stuttering with some phones

aaraku05, 2024-11-01:
This whole thing is working seamlessly with my s23 ultra 

But is stuttering and laggy with my OnePlus Nord CE 2 5g 
 Pls address this issue and provide a solution

shivammdi91, 2024-11-07:
I am also facing similar issue in OnePlus Nord CE 2 5g
Please suggest How can improve the performance.
I have also tried raspberry official cable.

aaraku05, 2024-11-10:
> Which Raspberry Pi board are you using?
> 
> Some people are experiencing issues because of the 2.4GHz wifi on RPi Zeros. RPi 3A+ or RPi 4 with 5GHz wifi might have better results in some cases, in case you can try that out.

Pi 02w

puregreen59, 2024-11-14:
I've also got some stuttering only on audio, it's mostly fine but I'd say every 5 - 25 seconds it stutters 

Phone: OnePlus 11 5G

yuppicide, 2024-11-16:
Pixel 9 Pro XL with RPI4B. Stuttered once or twice. Tonight is my first time trying it

steinhuegel, 2024-11-17:
Same here. Stuttering with oneplus 8 on 2w

Bala3007, 2025-01-21:
Im also facing stuttering only on audio with rpi 02w.
Phone: oppo Reno 8 5g

Luen, 2025-01-22:
Have a try of @manio 's build with the https://github.com/manio/aa-proxy-rs rust implementation. Seems to have less stuttering for me but still testing it.

manio, 2025-01-23:
@Luen
let us know if it is better over longer testing time period... Wondering if it also helps @banzonam in #85 or if it's unrelated...

banzonam, 2025-01-26:
Well, I'm having a lot of troubles all around. I tried v0.3.0 of the rust implementation, I THINK. I flashed the image and ran it on my Pi0w, but couldn't even find the device on the list of BT devices for initial connection. I saw its Wifi connection pop-up, but that's it. I went back to the v0.4.1 version and couldn't get AA to show up on either of car#1 or car#2's the stereos, despite the notification saying "AA connected to your car". AA had updated to a newer version 13.4.645054 between my last post and now (versus v13.3.644654 when it was working last). I went back to my AAWireless adapters to maintain voice functionality, both still work in both cars with the newer AA version. So, not sure what's going on. I'm not a dev at all, just a user and a professional test engineer for work.

manio, 2025-01-26:
@banzonam It's a known (not yet fixed [issue](https://github.com/manio/aa-proxy-rs/issues/3)) with `pi zero w`, I'll just prepare a full image for you...

shivammdi91, 2025-01-26:
https://github.com/manio/aa-proxy-rs
Anyone has tried this image with OnePlus Nord phones.
I am facing too much lags and disconnecting issue

banzonam, 2025-01-26:
> [@banzonam](https://github.com/banzonam) It's a known (not yet fixed [issue](https://github.com/manio/aa-proxy-rs/issues/3)) with `pi zero w`, I'll just prepare a full image for you...
> 
> Edit: here you are: [raspberrypi0w-sdcard.img.xz](https://skyboo.net/temp/aa/raspberrypi0w-sdcard.img.xz)

Got it thank you. I still think there's something wrong on my end. I downgraded to my previous version of AA, put the v0.4.1 on the card, and I still can't get AA to show up on my stereo. I also tried the image you posted, but same result. Notifications show Connected, but also Looking for Android Auto at the same time. So, until I can get that figured out, I don't think I can help much. Sorry, all!

manio, 2025-01-26:
> Edit: I was able to see and connect to this new image's BT, but again, AA never displayed on my stereo.

Try to forget all devices (in phone, in car). Also you can try to connect phone directly to car with using a wire beforehand.

banzonam, 2025-01-26:
Figured it out, bad cable, my bad. So, with v0.3.0 I still have poor voice command performance. I double checked with both AA versions v13.4.645054 and v13.3.644654. In the same config, my AAWireless adapter has expected voice command performance. Thanks.

manio, 2025-01-27:
Glad you figured it out!

aaraku05, 2025-02-02:
By the letest aa update by Google the issue is resolved on the main nisargs project 😃😃

JohnG2929, 2025-02-04:
Just set this up on a Pi Zero W.  It works amazing in my 2016 Silverado, however with the same cable etc, in my 2018 volvo XC90 there is "stuttering" during music playback. I would not contribute this to a hardware( pi/cable) issue, rather a buffer or communication speed setting that may be different between interfaces used by both manufacturers.  Is there a way to try and tune this?

DecrepitHuman, 2025-03-25:
Hey guys, I have the same issue with the latest version running on a brand new Pi Zero 2W. I am using it with my 2023 Chery Tiggo 4 Pro and it's very inconsistent. I will have no problems for weeks but randomly it will stutter rapidly (Consistently stutters every second for a like 100ms and then it's fine again for a second and stutters for another 100ms) which is extremely annoying and can only be fixed if I unplug the Pi and plug it back in. I haven't tried the Rust port yet but will do so shortly, just thought I should leave a comment.

Tttyagi123, 2025-06-05:
Try to downgrade output of source player to normal if set higher like spotify audi output or poweramp output and check if it reduces stuttering

kadimgh, 2025-07-22:
> Try to downgrade output of source player to normal if set higher like spotify audi output or poweramp output and check if it reduces stuttering

@Tttyagi123 
Do you mean to downgrade the quality of audio stream to avoid stuttering? (For instance in Spotify for Android it is under Settings and Privacy > Media quality > Cellular streaming quality)

Tttyagi123, 2025-07-22:
> [@Tttyagi123](https://github.com/Tttyagi123) Do you mean to downgrade the quality of audio stream to avoid stuttering? (For instance in Spotify for Android it is under Settings and Privacy > Media [...]

Na, i tried it and it din work. I think it has to do with 2.4g network by 02w, i have switched to pi 3a plus now which has a 5g network and the music is lag free now.

kadimgh, 2025-07-23:
> @kadimgh <https://github.com/kadimgh> do you also have this problem with
> aa-proxy-rs <https://github.com/manio/aa-proxy-rs/releases>?

Let me try it tomorrow, I was assuming I'd need to sftp the binary to the
RPi, but it looks like there are ready-to-use images with aa-proxy-rs
included

kadimgh, 2025-07-23:
> [@kadimgh](https://github.com/kadimgh) do you also have this problem with [aa-proxy-rs](https://github.com/manio/aa-proxy-rs/releases)?

I've tried aa-proxy-rs 0.7.0 and I'm facing the same stuttering issues as with aawgd. Let me know if there is any data helpful in the logs to debug further

kadimgh, 2025-07-23:
My current setup out-of-the-box for aa-proxy-rs 0.7.0 is:

'''
hw_mode=g
#ieee80211n=1
channel=1
'''

I was (wrongly) assuming it would use 5G radio on RPi 4 by default. I can certainly tinker with these settings if it could improve performance, I'll start with the ones recommended by you and then see how it goes

kadimgh, 2025-07-23:
I've found out that ```hostapd``` is using ```/var/run/hostapd.conf```, which is in turn generated based on ```/etc/hostapd.conf**.in**``` that has the 5Gh configuration in it for RPi4. I'll try to modify it to use either a non-default channel or 2.4GHz and see if it improves the performance

manio, 2025-07-25:
> [@manio](https://github.com/manio) I've tried several different configurations of WiFi (5GHz using channels 36 and 48, 2.4GHz using channels 1 and 2) and I'm getting audio dropouts with all the configurations I've tried. The pattern of dropouts changes depending on WiFi settings, but I was not able to eliminate them or bring down to an acceptable level.

:(

> Briefly looking at the `aawgd` code it looks like the packets are forwarded [one by one](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/package/aawg/src/proxyHandler.cpp#L120) as soon as they arrive. I can't read Rust well, but I assume that `aa-proxy-rs` is doing the same thing.

correct

> Were there any attempts to implement small buffer in the dongle to mitigate the jitter and dropouts?

No, because the latency is the key. It has to be the lowest. All the buffering, etc is done on the phone and HU internals and the low-latency streaming protocols. I cannot buffer data because it would be worst, and make more disconnects.

> If buffering is not feasible or compatible with Android Auto architecture then are there any other known mitigations (for instance Android Auto on the phone side allows to configure audio codec and "Prefer AAC-LC over PCM", not sure though whether the head unit will respect this preference and if AAC is any better)? Also aa-proxy-rs has an option to disable Media/Audio sink, does it by any chance force the phone to route audio over BT instead of AA?

Those option could be helpful depending on your needs. It was prepared eg for the scenario when a passenger is using a phone for eg youtube/playing games and AA is used only for navigation and the music has to be _NOT_ routed via AA, etc...
In this case probably it could be possible to send via bluetooth.... No idea... you'd nee to test those options...

If you like I can also try to prepare a 64bit optimized RPI4 image for you (it is not yet official but we are testing those builds and I think it should work). Maybe it would work better...?

manio, 2025-07-25:
> If you like I can also try to prepare a 64bit optimized RPI4 image for you (it is not yet official but we are testing those builds and I think it should work). Maybe it would work better...?

Here you are in case you want to give it a try:
https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi4.img

Tttyagi123, 2025-07-25:
> > If you like I can also try to prepare a 64bit optimized RPI4 image for you (it is not yet official but we are testing those builds and I think it should work). Maybe it would work better...?
> 
> Here you are in case you want to give it a try: https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi4.img

How bout one for pi3a as well, will love to test it too ;)

kadimgh, 2025-07-25:
@manio 
> No, because the latency is the key. It has to be the lowest. All the buffering, etc is done on the phone and HU internals and the low-latency streaming protocols. I cannot buffer data because it would be worst, and make more disconnects.

Just to close this topic: is this an assumption that small buffer (like 2-4 packets) will lead to more disconnects or was it verified in practice?

For instance I can speculate that HU that implement AA Wireless do _additional_ buffering to mitigate dropouts compared to HU that implement _only_ AA Wired and have a reliable time guarantees. If that was the case then the responsibility for buffering would fall on the dongle in case of HUs with _only_ wired AA support

> If you like I can also try to prepare a 64bit optimized RPI4 image for you (it is not yet official but we are testing those builds and I think it should work). Maybe it would work better...?

I'll try both 64bit image and disabling audio sink over the next few days, thank you for the test image

manio, 2025-07-25:
@kadimgh
> Just to close this topic: is this an assumption that small buffer (like 2-4 packets) will lead to more disconnects or was it verified in practice?
>
> For instance I can speculate that HU that [...]

Just to clarify - I’m not an expert, but based on my experience with Android Auto and working with the protocol (I’m doing a MITM setup in my project, so I’ve had to dive into this a bit), the key thing here is low latency.
What I was trying to explain is that AA is made up of several layers. Right now, we’re talking about the network layer where the data is actually sent. But there are also upper layers, like the streaming codecs, that kick in when the connection quality drops - they try to adapt to poor throughput, but without increasing latency.
It’s kind of like a Skype call - if the connection gets bad, it doesn't just start buffering more; instead, the audio/video quality drops, but it keeps things in sync. Same goes for using a Bluetooth headset - the quality may degrade, but keeping it real-time is more important.
If I try to add any extra buffering, the phone picks up on that and disconnects. It seems to actively monitor latency, and if it gets too high, it can just drop the whole connection. So buffering more isn't a fix - it actually makes the situation worse. That’s the point I was trying to make.
Even just logging the full AA stream to the SD card on a Raspberry Pi caused issues - and that’s just from the added I/O overhead. It was enough to trigger disconnects.

@Tttyagi123
> How bout one for pi3a as well, will love to test it too ;)

https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi3a.img
Let me know how it works and if you have problems... it would be valuable for me...

Tttyagi123, 2025-07-25:
> [@kadimgh](https://github.com/kadimgh)
>
>
> Just to clarify - I’m not an expert, but based on my experience with Android Auto and working with the protocol (I’m doing a MITM setup in my project, so [...]

Sure will

introkun, 2025-07-27:
Same here. Stuttering with Sasmugn S22 Ultra on 2w

kadimgh, 2025-07-30:
> I assume you're using 5GHz WiFi? Did you try to [change channel](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/245)?

With the 64 bit image I've only tried the default WiFi config (which I assume is 5GHz channel 36). I've tried other options [before](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/217#issuecomment-3115084007) when running 32 bit image

I don't know if it was a coincidence, but with 64 bit image I often needed to re-plug RPi to get the AA to work, so I returned back to 32 bit image, which connects reliably 9 out of 10 times when I start the car

manio, 2025-07-30:
> With the 64 bit image I've only tried the default WiFi config (which I assume is 5GHz channel 36). I've tried other options [before](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/217#issuecomment-3115084007) when running 32 bit image

yes

> I don't know if it was a coincidence, but with 64 bit image I often needed to re-plug RPi to get the AA to work, so I returned back to 32 bit image, which connects reliably 9 out of 10 times when I start the car

I understand, it's a shame you don't have any logs from those issues... I'd really like to take a look at them. I care about making sure this works properly...

kadimgh, 2025-07-30:
> I understand, it's a shame you don't have any logs from those issues... I'd really like to take a look at them. I care about making sure this works properly...

I can re-flash 64 bit image and collect the logs. Please provide instructions. (I've tried navigating to http[s]://10.0.0.1/ in Chrome from either laptop or phone connected to the dongle's WiFi network but I did not see a config page)

manio, 2025-07-30:
Sure... but hold on I'll build the fresh image for you (there was some changes recently)...

manio, 2025-07-30:
> I only have RPi**4**, will this build work for me?

No, for Pi4 use this:
https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi4.img

> I'm not sure what this means. I've tried to navigate to http://10.0.0.1/ from both the phone and laptop that were connected to AA WiFi network. For instance the laptop can do SSH to 10.0.0.1, but cannot do HTTP. I can also modify the config file via SSH instead of web interface if needed

There was a bug which leads to aa-proxy-rs exits when there's no HU connected. It's now fixes so webserver should be available.

introkun, 2025-07-31:
> Same here. Stuttering with Samsugn S22 Ultra on 2w on a standard image from the latest release

It's way better on aa-proxy image.

kadimgh, 2025-07-31:
@manio 
> No, for Pi4 use this: https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi4.img

I've flashed it and enabled persistent logging (with some difficulties described at the bottom) . I'll test it tomorrow. Please confirm this is 64-bit image, because it's <150MB, while the previous one was ~190MB

As for enabling logging using the web-interface there are two issues, so I ended up enabling it by editing the ```config.toml``` via SSH:

1. "Save"->"Apply/Restart" sequence seems broken, at least when used in Chrome. See the screenshot
2. ```/persist/aa-proxy-rs.log``` example could be confusing because there is no ```/persist```. Please consider updating it to something like ```/data/aa-proxy-rs.log```

[image]

manio, 2025-07-31:
> I've flashed it and enabled persistent logging (with some difficulties described at the bottom) . I'll test it tomorrow. Please confirm this is 64-bit image, because it's <150MB, while the previous one was ~190MB

That's correct. It's because we are still working on this. The root filesystem is readonly and the `/data` is making for overlay. You can make the last partition bigger to fill your entire SD card if you need more space.

ad1: thank you for the report I have to see what's wrong, it was probably broken after some recent change...
ad2: you're right I'll change that as well (we decided about that `/data` recently so this was not yet covered in the docs)

manio, 2025-07-31:
@kadimgh when you'll test/collect the logs, please also try to set `bt_timeout` to eg 3600 (very high) and see if it is better.

rickgitdone, 2025-07-31:
is there a newer image with the fixes for pi 2W ?

rickgitdone, 2025-07-31:
if you don't mind to disable wpa_supplicant service as well .. to ensure it doesn't have a change to make any interference

manio, 2025-07-31:
ok, regenerated with these fixes, `wpa_supplicant` is not even starting, so don't bother at all...
https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi02w.img

jeremywmoore, 2025-08-03:
> ok, regenerated with these fixes, `wpa_supplicant` is not even starting, so don't bother at all... https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi02w.img

I was having some intermittent stuttering issues in one of my cars (2022 Toyota Sienna) with the 0.5.0 release and a quick test with this build seems to have resolved it. FWIW I don't have any stuttering issues on 0.5.0 in my 2018 Nissan Leaf (maybe the lower resolution of the display puts less demand on the rpi02w?).

ekutner, 2025-08-05:
> ok, regenerated with these fixes, `wpa_supplicant` is not even starting, so don't bother at all... https://skyboo.net/temp/aa/aaproxy-br/sdcard-pi02w.img

I tried with this image and I'm sorry to say that for me it didn't solve the problem. RPi Zero 2 W, Skoda Kodiaq 2019 and Samsung Note 10+

manio, 2025-08-05:
> I tried with this image and I'm sorry to say that for me it didn't solve the problem. RPi Zero 2 W, Skoda Kodiaq 2019 and Samsung Note 10+

Sorry to hear that. Can you send me the logs when this happen?

> Just to be sure I also tried with 0.7.0 and sure enough it has the same problem, however, while testing it I noticed that there is a very high correlation between those "stuttering" episodes and getting a toast message saying "Couldn't connect to Skoda BT 0764", which is the ID of my car's built-in bluetooth.

Is this toast on your phone? And while AA is working?

> I don't know if it's normal for my phone to try to connect to the car's bluetooth all the time and fail but I suspect it is somehow behind this stuttering issue.

A normal (clean/fresh) procedure is:
1. Connect to your car BT - for audio calls, phonebook etc - your phone and car are paired and profiles saved
2. Connect the RPI dongle to USB port
3. From your phone connect to the bluetooth of the raspberry - AA handshake would start
4. Your phone will connect to a WiFi of the dongle and bluetooth will be disconnected
5. Your phone will connect to you *CAR* bluetooth (for voice calls) keeping WiFi for AndroidAuto at the same time

ekutner, 2025-08-05:
> Is this toast on your phone? And while AA is working?

Yes 

> A normal (clean/fresh) procedure is:
>
>     1. Connect to your car BT - for audio calls, phonebook etc - your phone and car are paired and profiles saved
>
>     2. Connect the RPI dongle to USB [...]

Thank you for this explanation. I unpaired my phone and car on both ends end did the whole process again. After some hiccups I got everything working and behaving as you described. After doing this, for the first time, as long as the phone screen was off, I had no stuttering on the audio. The fact that the phone screen on/off affects this and also the attempts to connect to the car bluetooth before the rest, suggest that phone performance and load might be a factor here. My phone is 5 years old but I will try with my wife's Samsung Z6 later today and see how it behaves.

Thanks again for your help and great work on this!

DecrepitHuman, 2025-08-12:
I have precisely the same stuttering issue and it is indeed very inconsistent as the conversion discussed the WiFi channels and such. Sometimes it is "a bad day" and I'll have the issue two or three times in a row with just a few minutes delay between them - and then the stuttering gets so bad that I can't skip songs, go to settings, open apps etc and I have to physically unplug the Pi and plug it back in for it to regain functionality.

I am running on a relatively new Zero 2W but my images are likely a few months old - I think I was one of the 1X commentors on this thread that kind of kick-started everybody looking into the problem so I will also just check that I am running the latest versions. Haven't changed anything at all config wise so maybe I should also figure out how to ensure it's running on 5Ghz and not 2.4Ghz and so on

manio, 2025-08-13:
> I wrote Pi 2 W image 0.80 to sdcard and am unable to change my Country code since the partition has zero space to allow me to make a change on /etc/aawgd.conf ??

I plan to add this option to web interface. How you was changing this? Externally from linux or from running SSH?

rickgitdone, 2025-08-13:
> > I wrote Pi 2 W image 0.80 to sdcard and am unable to change my Country code since the partition has zero space to allow me to make a change on /etc/aawgd.conf ??
> 
> I plan to add this option to web interface. How you was changing this? Externally from linux or from running SSH?

I change using Linux mounting the SDCard ... just leave a few megs at the end of the partition ... like the other images originally had

Tttyagi123, 2025-08-13:
> > Zero2W is not supporting 5GHz. I can recommend for you:
> > 
> > * https://github.com/aa-proxy/aa-proxy-rs/releases/tag/v0.8.0
> > * https://skyboo.net/temp/aa/joka90/rpi02w_sdcard.img
> > 
> > And provide a feedback regarding stuttering to us...
> 
> Downloaded 0.8.0 from stable build release repository, will try it in some time. Thanks.

@manio i installed it today finally, will yest it tomorrow,it connected right away though. Oke thing i noticed is, this build does not provide an option to change the config file like in the 0.5.0 where i wud set the hdmi scan off, set country code, etc.is it deliberate or something which is missed?

rickgitdone, 2025-08-14:
> > is it deliberate or something which is missed?
> 
> as you can read above... it is because I plan to add this settings int the [webinterface](https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#embedded-web-interface). For now you can edit `/etc/aawgd.conf` on second (rootfs) partition

when this idea is ready to test so us consumers can attempt to use/adjust the now locked down image let us know.. thx

manio, 2025-08-15:
> I would also ask we keep aa-proxy talk on github aa-proxy/aa-proxy-rs repo.. there is constant cross-over which confuses nisargjhaveri/WirelessAndroidAutoDongle with aa-proxy issues, code and functionality

You're right, for the most part. While my project did originally rely on this one’s base, that’s no longer really the case now that we have our own Buildroot. I had this project `Watched` and simply tried to be helpful by responding to user issues when I could. Sometimes I offered help without mentioning `aa-proxy`, but admittedly, I did often end up suggesting people try my project...

That said, I really don’t want to come across as pushy, and you’re right - I’ll stop doing that going forward.

> when this idea is ready to test so us consumers can attempt to use/adjust the now locked down image let us know.. thx

Regarding the first part of your message: I wouldn’t really call users “consumers.” That sounds more like something from a commercial project, where people pay and expect certain functionality. This is an open-source project with more of a community of enthusiasts and users. No one’s paying me - I work on this in my free time - so not everything is immediately available. It takes time and effort to reach the level of functionality that you're expecting to be out-of-the-box - like setting WiFi country code (which in fact can still be set manually in config file - exactly as in **this** project and not "locked down").

Tttyagi123, 2025-08-19:
i have been on the 64 bit build for a week and  havent noticed any issues on my pi3a so far, all good :)

Tttyagi123, 2025-08-19:
> You're right, for the most part. While my project did originally rely on this one’s base, that’s no longer really the case now that we have our own Buildroot. I had this project `Watched` and simply [...]

i vouch for him, he is the one guy who spent half of his day trying to help us get our software working relentlessly and in return for what? nothing to be precise! is he obligated to do it??? No!

i think the least we call can do is be thankful for people like him who "help" us without asking for anything in return. 

btw, he is the major contriutor in the main project too so it really does not matter if he is asking to try his build and there is nothing wrong in that, we always have the freedom not to. that being said, i love to try new things so i enjoy trying  any new build he makes and provide him inputs so that he can make it better, we feed the technical monster in him so that he works his arse off to give us the more clean and bug free builds by providing the fedback :P :D U go My man, create magic like u always do!!!! lol

DecrepitHuman, 2025-08-27:
Hey guys, just a follow-up on my testing of the optimized zero2w image.. I have not once experienced the stuttering but the problem is that half the time it simply doesn't connect to my phone. I have to occasionally spam connect to either the WiFi or through bluetooth for it to even remotely try to work. Once connected it seems fine but has honestly made it more frustrating.

At the moment I am considering upgrading to a Pi 4 2GB board or something like that.. I assume it does have official support including 5Ghz which you have suggested as the solution to most lag related issues correct?