nisargjhaveri/WirelessAndroidAutoDongle #219: [ANNOUNCEMENT] `aawgd` Rust rewrite

manio, 2024-11-02:
Nisarg,
I hope you don't mind that I am creating this special "issue" here but as my project is strictly related and dependent on your **WirelessAndroidAutoDongle** so I am posting it here.

So to the point: I rewrited `aawg` in Rust and I hope someone besides me will also find it useful, so I am providing a link to my new project: [aa-proxy-rs](https://github.com/manio/aa-proxy-rs)
There is still left some of work to do (details in [README](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#-aa-proxy-rs)), but it is now working for me so I am sharing this info.

I am also CC-ing other ppl involved somehow in this project, maybe you guys find my work useful...
CC:
@BluemediaGER @hkfuertes @KadenK @Thom-x @Demon000 @Ioniq3 @NabeelUppel @ankush-ksharma @krish-ag @mevivek

regards,
manio

closed 2024-11-03 by manio

nisargjhaveri, 2024-11-05:
@manio, Thanks a lot for sharing this! This is exciting. Even though I'm new to Rust, might be useful to move to it eventually given the benefits it provides over C++ implementation.

Until now, I was not treating aawg as replaceable and the interaction with the rest of the system might not have been stable. I'll try to keep the interface with the rest of the setup stable enough or will try notify you of any major changes. Feel free to suggest any changes if it helps with better integration overall.

manio, 2024-11-06:
Great, thank you very much! :)