nisargjhaveri/WirelessAndroidAutoDongle #221: Feature Request

hitensam, 2024-11-05:
Once again, thanks @nisargjhaveri for creating such an amazing solution. It works fine with only a few minor glitches in the connection, but it automatically reconnects after a minute or two.

Actually, I wanted to change the names of the Bluetooth and Wi-Fi, as we are using this approach in multiple vehicles, thanks to you.

For another vehicle to work with the Bluetooth connection, we first need to unpair and then pair it again.

Could you kindly let me know all the files where these changes should be made, so everything works perfectly?

Also, if possible, can we create a random Wi-Fi password for each session? What happens currently is that if we are using a phone that was previously connected to Android Auto, it connects to the SSID before the device we actually want to connect. My suggestion is that when the phone initially pairs with a device, the new password is shared with the phone, allowing it to connect smoothly. Then, after the next reboot, a new password is generated.

Additionally, I would like to set a limit on Bluetooth connections. If a device has already established a connection with the Raspberry Pi and is working with the head unit, no other device should be able to pair. Pairing should only happen during the initial setup or if the connection is lost. I think this would be a step forward in creating a better product.

Let me know how I can contribute in any way!

nisargjhaveri, 2024-11-05:
With 6aecca66427cb11dfeaa77f82366570ba3130956, we support changing the Bluetooth name. In fact, by default it will use a unique bluetooth name.

#200 makes an attempt to create random password for wifi. You can try that out and see if that helps? It'll help if more people can test it out.

For limiting the bluetooth connection, depending on the connection strategy, it might already be disabled. Are you seeing any issue with having the pairing on after the connection? I'd like to understand more on this.

hitensam, 2024-11-05:
As of now, I only see this name on all my devices (Raspberry Pi):

- Bluetooth
    [image]

- Wi-Fi
    [image]

Because of the same Bluetooth name, I need to unpair and then pair the devices again.

For issue #200 and testing the pairing after connection, I will try it and let you know after some time, as I currently do not have the device with me.