nisargjhaveri/WirelessAndroidAutoDongle #222: Audio Interferace/Wifi Disconnected when Braking/Engine Off

yaagesh, 2024-11-08:
Hi, Thanks for making this wonderful solution. But I'm facing minor issues with my Zero 2W implementation.

I have connected board to my Tata Punch 2021 variant and the solution works great while driving for past 2 days. But When I'm braking or when the vehicle in the low speed there is a audio interferance. And while the Engine is turned off I notice teh board reboots sometime or the same audio interferance is frequesnt.

I'm thinking that there may be a voltage change during the above scenarios, I tried to give supply through the DashCam Charging adapter but the Pi the returning power to the adapter and powering the DashCam (when the DashCam power port is off and DashCam power and Pi Power is in same adapter). Does anyone facing the same issue or anyone found a solution?

manio, 2024-11-19:
Look here: #211

vintan, 2024-12-27:
I am using WAA latest (or at least I can say that it is downloaded in November) with RPi Zero 2 W. I am using it with Tata Altroz XT 2023 and Moto Edge 50 Pro.

I too noticed the issue you mentioned. I am still in the process of identifying when exactly the audio stutter. But it is seen when braking or sudden slowing down of the vehicle or sometimes when engine jerks before down shifting the gear.

The behaviour is that the map audio or the music will kind of buffers or skips few frames, something like that.

If it is power related then may be I will have to try the solution mentioned in #211 or similar.