nisargjhaveri/WirelessAndroidAutoDongle #223: Capture Android Auto Start Request

NabeelUppel, 2024-11-08:
Hi @nisargjhaveri 

I was debugging why it doesn't work with car despite it working fine in another and I came to conclusion that my headunit doesn't seem to send the accessory start request 

I disabled the start script located at etc/lib.d/93_aawgd (just renamed aawgd to aagwd) and ran the binary locally in usr/bin and my phone would connect to bluetooth and the TCP server would start, I would get the connected to android auto on my phone but it seems as though my headunit has no idea that it's connected. Then it would timeout and try again. 

When I plug in my phone via cable it works perfectly fine, so I figured that the problem lies between how the pi is communicating to the headunit.

I've tried the USB connection strategy mode and the Bluetooth is never turned on so I think my headunit is expecting something specific which I cannot figure out and that brings me to my point.

Is it possible to capture the handshake between my phone and headunit when connected via cable?

Phone: S22+
Car: Mazda 3 2015
Pi: Zero 2 W

NabeelUppel, 2024-11-08:
Another thought I had which I'm not sure if possible and I'm not sure where to start with this is to emulate android usb.

As in using the otg port on the pi, make it so the headunit believes that a wired android is connected to it. 

I think this is what AAWireless Dongle also makes use of calling it passthrough. 

I have no idea how this software works low level, so if this is not possible then just say so.

NabeelUppel, 2024-11-08:
Okay I think I got it working again!

I forked main, added some debug logs and used the workflow to build it. 

I then factory reset my headunit.

Completely removed the bluetooth, WiFi, uninstalled AA updates via the play store from my phone..

Set it up android auto via my phone and bang its working again. No idea which step made a difference.

nisargjhaveri, 2024-11-11:
Hi, @NabeelUppel! Glad that you got it working. There are some reports that suggest removing wifi/bt and Android Auto storage on phone helps or resetting headunit also helps sometimes for other dongle as well. Not really sure why though and what can be done to prevent the need?

> Is it possible to capture the handshake between my phone and headunit when connected via cable?

It might be possible with some usb MITM hardware. It might also be possible with a Raspberry Pi 4. But in general, I found it a lot more involved. Even if we get the raw data transfer, we need to understand the protocol. It is using and try to replicate that.

> Another thought I had which I'm not sure if possible and I'm not sure where to start with this is to emulate android usb.

We essentially do this. We present RPi as a MTP device with Pixel's usb vid/pid. The aim is to make headunit think a Android Phone is connected. The MTP implementation itself might be different, and we should see if there is a scope of improvement.

NabeelUppel, 2024-11-12:
> Hi, @NabeelUppel! Glad that you got it working. There are some reports that suggest removing wifi/bt and Android Auto storage on phone helps or resetting headunit also helps sometimes for other dongle as well. Not really sure why though and what can be done to prevent the need?

I'm also not sure, it seems to be working consistently now, I did manage to run it awhile back for a month or two and then it stopped. I'm sure you aware of this with the amount of issues I've opened up haha. 
If it stops working I'll trying clearing the bluetooth devices from my car before going nuclear and resetting. 

> It might be possible with some usb MITM hardware. It might also be possible with a Raspberry Pi 4. But in general, I found it a lot more involved. Even if we get the raw data transfer, we need to [...]

What does the Pi4 have that makes it possible? I also tried looking into it and it seems quite obfuscated and didn't find much. 

> We essentially do this. We present RPi as a MTP device with Pixel's usb vid/pid. The aim is to make headunit think a Android Phone is connected. The MTP implementation itself might be different, and [...]

Again man, I have to say great work on this it's super impressive. 

Feel free to close this issue.

nisargjhaveri, 2024-12-05:
> What does the Pi4 have that makes it possible?

Pi 4 can simultaneously work as usb device using the usb-c port and usb host using the other usb ports. This makes it possible to simply transfer all the traffic between the connect device and host and behave as a usb proxy device. You don't necessarily need Pi 4 for this, I belive bunch of other devices can also do this.

I did manage to use this once. But as I said earlier, it might be difficult to get much useful insights this way. https://github.com/AristoChen/usb-proxy

I'll close this one for now. Feel free to reopen!

closed 2024-12-05 by nisargjhaveri

manio, 2025-07-27:
From I've read @NabeelUppel's code in the fork he is only delaying start by specific number of seconds... :)

NabeelUppel, 2025-07-27:
Hi @justaperson270 

What worked for me was using a good USB cable with data capabilities and now and again I have issues where it still won't connect which I resolve by resetting the head unit. 

There isn't much to set up anyway and it happens very rarely.

justaperson270, 2025-07-28:
> Hi [@justaperson270](https://github.com/justaperson270)
> 
> What worked for me was using a good USB cable with data capabilities and now and again I have issues where it still won't connect which I resolve by resetting the head unit.
> 
> There isn't much to set up anyway and it happens very rarely.

Ok, Thanks for the input!

Is there a solution to my problem, @manio  or @nisargjhaveri ? As I mentioned, the adapter works perfectly in other cars with wired Android Auto, but it doesn’t seem to function in my car for some reason. Could it be an issue with my car's infotainment system or Bluetooth? I can send my logs if necessary, but I'm still learning how to do that, so any guidance would be appreciated since I'm not very technically advanced.

justaperson270, 2025-07-29:
> If you're not advanced you can try to flash [this](https://github.com/aa-proxy/aa-proxy-rs/releases) and it's then easier to obtain the logs via [webinterface](https://github.com/aa-proxy/aa-proxy-rs?tab=readme-ov-file#embedded-web-interface). Click download logs and we could see what could be wrong...

Hi @manio, here are my logs. Some of them might be different because I tried to download the logs right when the problem occurred before the adapter reset itself (so the log will be different depeding on the file)

[20250728234704_aa-proxy-rs (1).log](https://github.com/user-attachments/files/21481387/20250728234704_aa-proxy-rs.1.log)
[20250728235127_aa-proxy-rs(2).log](https://github.com/user-attachments/files/21481389/20250728235127_aa-proxy-rs.2.log)
[20250728233455_aa-proxy-rs.log](https://github.com/user-attachments/files/21481388/20250728233455_aa-proxy-rs.log)

manio, 2025-07-29:
Reviewed your logs:
First is the most advanced connection: the bluetooth handshake was done, your phone connected to raspberry and it is opening connection to headunit, unfortunately the log stops there so I don't know what was next...
Second logs stops on waiting for phone connection
Third stopped in the middle of bluetooth handshake...

Can you just give it more time to see what is wrong? Please also:
1. change USB cable/make sure it is data cable (not for charging only)
2. try to disable/enable legacy mode of the USB in webinterface

is it working OK when you connect your phone directly to the headunit?

justaperson270, 2025-07-29:
Hi @manio,
Here are my new logs after leaving the adapter in the USB for longer:

[20250729140048_aa-proxy-rs(attempt 1).log](https://github.com/user-attachments/files/21496765/20250729140048_aa-proxy-rs.attempt.1.log)

[20250729140603_aa-proxy-rs(attempt 2).log](https://github.com/user-attachments/files/21496768/20250729140603_aa-proxy-rs.attempt.2.log)

[20250729141203_aa-proxy-rs(after legacy #1).log](https://github.com/user-attachments/files/21496769/20250729141203_aa-proxy-rs.after.legacy.1.log)

[20250729141602_aa-proxy-rs(after legacy #2).log](https://github.com/user-attachments/files/21496770/20250729141602_aa-proxy-rs.after.legacy.2.log)

[20250729142340_aa-proxy-rs(after modifying timeout time legacy mode on).log](https://github.com/user-attachments/files/21496772/20250729142340_aa-proxy-rs.after.modifying.timeout.time.legacy.mode.on.log)

[20250729150710_aa-proxy-rs(after modifying timeout time legacy mode off).log](https://github.com/user-attachments/files/21496773/20250729150710_aa-proxy-rs.after.modifying.timeout.time.legacy.mode.off.log)

[20250729145513_aa-proxy-rs(Car that works with adapter for reference).log](https://github.com/user-attachments/files/21496776/20250729145513_aa-proxy-rs.Car.that.works.with.adapter.for.reference.log)

manio, 2025-07-30:
Reviewed all logs... The main problem is that you are not connecting to the dongle from your phone for the first time. You have to. Otherwise the raspberry doesn't know your phone BT address and cannot start communication. The reference log is the one which seems ok (not counting the connection time on phone side which was 49 secs!). The last log also seems fine but you said it resets.

Try again the with legacy mode=OFF, but then also connect to the BT dongle from the phone and give it some more time - as we can see that even 49 second your android was trying to connect... which is very long.

justaperson270, 2025-08-09:
Hi, sorry for the delayed response, but I finally got it working. I was about to give up, but turns out my car likes to have the SD card formatted into FAT32 (which the USB was not formatted to). I reformatted the SD card to FAT32, then flashed the SD card again with the files, and everything worked out!

manio, 2025-08-09:
Only for your information: whatever you did on your card before flashing doesn't matter because flashing is overwriting it. Besides the flash card is important only for raspberry, not to your car.
Nevertheless, I am glad that it's working!