nisargjhaveri/WirelessAndroidAutoDongle #226: Is it possible to be run on same device as Crankshaft/OpenAuto Pro?

KreAch3R, 2024-11-12:
Hello @NabeelUppel, 

I just found this repo and I am amazed by your code and your will to open source it! 

I am running OpenAuto Pro (a closed source Crankshaft fork) on a raspberry pi 4 in my car for some years now, and two months ago the integrated wireless Android Auto way stopped working for Android Auto apk >12.6. Something changed on the way the app is asking for wifi handshake. Crankshaft experienced the same problem, although they didn't even have a working AA wireless.

Now, OpenAutoPro went out of business and they won't be updating the app or sharing the code, so I'm just stuck with the outdated code. 
Then I found this repo and it got me thinking, instead of connecting a rpi-zero to my rpi-4, is there any way that I can install the necessary code on the rpi-4 itself, and then make AAWG connect to OpenAuto running on the same machine, using the USB Accessory Mode?
This is way over my head, but the only way I can imagine that this can work is by emulating a USB connection.

Do you have any ideas? Can I make it happen somehow? Or should I install a separate raspberry pi alongside my first. Which would be a shame but that's what close-sourcing does. :(

Again, thank you for your code. This isn't an issue, but I thought it belonged here instead of reaching out by email.

NabeelUppel, 2024-11-12:
Hi!
I'm far from a contributor. 
The guy you looking for is @nisargjhaveri, this is his project and he deserves all the praise.

KreAch3R, 2024-12-02:
@nisargjhaveri Any comment on this?

I did get a rpizero2w and it works great but connecting it to the rpi4 headunit, but I would still like to eliminate one board if possible.

I also get two different popup errors when connecting, which don't interfere with the usage:
`couldn't find matching udev device`

`unable to open MTP device "001,012"`

KreAch3R, 2024-12-22:
Well, after a couple of weeks of debugging and coding, I can confirm that **_it is_** possible to run `aawg` (specifically its re-write, `aa-proxy-rs`) in the same device as host and usb gadget, using `dummy_hcd` module as the UDC controller than handles the connection between the two sides. 

So I have a raspberry pi 4 running OpenAuto-Pro's custom Raspbian Buster image acting as the headunit host and in the same time, acting as the dongle using a statically built `aa-proxy-rs` and Android Auto Wireless works great. 

It does require rebuilding the kernel and adding some of the bluetooth and wifi hotspot configurations from this repo. 

I will upload all the necessary configuration in my repo here: https://github.com/KreAch3R/aa-proxy-oap

Thanks @nisargjhaveri and @manio !

```
2024-12-22, 22:22:30.185 [INFO] 🛸 aa-proxy-rs is starting, build: 20241222_212644, git: 20241222-4a21c16-dirty
2024-12-22, 22:22:30.243 [INFO]  main:  📜 Log file path: /var/log/aa-proxy-rs.log
2024-12-22, 22:22:30.243 [INFO]  main:  ⚙️ Showing transfer statistics: disabled
2024-12-22, 22:22:30.292 [INFO]  proxy:  🛰️ Starting TCP server...
2024-12-22, 22:22:30.292 [INFO]  proxy:  🛰️ TCP server bound to:  0.0.0.0:5288
2024-12-22, 22:22:30.292 [INFO]  proxy:  💤 waiting for bluetooth handshake...
2024-12-22, 22:22:30.317 [INFO]  usb:  Using UDC: "dummy_udc.0"
2024-12-22, 22:22:30.317 [INFO]  usb:  🔌 Initializing USB Manager
2024-12-22, 22:22:30.318 [INFO]  usb:  🔌 USB Manager: Disabled all USB gadgets
2024-12-22, 22:22:30.318 [INFO]  bluetooth:  🥏 Bluetooth alias: NaviPi
2024-12-22, 22:22:30.913 [INFO]  bluetooth:  🥏 Opened bluetooth adapter hci0 with address 00:1A:7D:DA:71:14
2024-12-22, 22:22:31.225 [INFO]  bluetooth:  📱 AA Wireless Profile: registered
2024-12-22, 22:22:31.226 [INFO]  bluetooth:  🎧 Headset Profile (HSP): registered
2024-12-22, 22:22:31.226 [INFO]  bluetooth:  ⏳ Waiting for phone to connect via bluetooth...
2024-12-22, 22:22:49.777 [INFO]  bluetooth:  📱 AA Wireless Profile: connect from: 30:74:67:AF:9C:C2
2024-12-22, 22:22:49.778 [INFO]  bluetooth:  📲 Sending parameters via bluetooth to phone...
2024-12-22, 22:22:49.778 [INFO]  bluetooth:  🛜 Sending Host IP Address: 10.0.0.1
2024-12-22, 22:22:49.778 [INFO]  bluetooth:  📨 stage #1 of 5: Sending WifiStartRequest frame to phone...
2024-12-22, 22:22:49.863 [INFO]  bluetooth:  📨 stage #2 of 5: Received WifiInfoRequest frame from phone (⏱️ 84 ms)
2024-12-22, 22:22:49.863 [INFO]  bluetooth:  🛜 Sending Host SSID and Password: NaviPi, AANaviPi
2024-12-22, 22:22:49.864 [INFO]  bluetooth:  📨 stage #3 of 5: Sending WifiInfoResponse frame to phone...
2024-12-22, 22:22:49.873 [INFO]  bluetooth:  📨 stage #4 of 5: Received WifiStartResponse frame from phone (⏱️ 9 ms)
2024-12-22, 22:22:54.922 [INFO]  bluetooth:  📨 stage #5 of 5: Received WifiConnectStatus frame from phone (⏱️ 5033 ms)
2024-12-22, 22:22:54.922 [INFO]  proxy:  🛰️ TCP server: listening for phone connection...
2024-12-22, 22:22:54.922 [INFO]  bluetooth:  🚀 Bluetooth launch sequence completed
2024-12-22, 22:22:55.143 [INFO]  usb:  🔌 USB Manager: Switched to accessory gadget
2024-12-22, 22:22:55.144 [INFO]  bluetooth:  🥷 Unregistering default agent
2024-12-22, 22:22:55.144 [INFO]  bluetooth:  📱 Removing AA profile
2024-12-22, 22:22:55.680 [INFO]  proxy:  📳 TCP server: new client connected: 10.0.0.20:36834
2024-12-22, 22:22:55.680 [INFO]  proxy:  📂 Opening USB accessory device: /dev/usb_accessory
2024-12-22, 22:22:55.681 [INFO]  proxy:  ♾️ Starting to proxy data between TCP and USB...
2024-12-22, 22:22:57.646 [WARN]  bluetooth:  🎧 Error waiting for HSP profile task: deadline has elapsed
2024-12-22, 22:22:57.827 [INFO]  bluetooth:  💤 Bluetooth adapter powered off
```

closed 2024-12-29 by KreAch3R