nisargjhaveri/WirelessAndroidAutoDongle #230: Not autoconnecting to Wifi when connected with car port!!!

tanmay380, 2024-11-30:
When i'm not connected to car and try to power on the PI by connecting it to a normal power source it'll first connect to bluettoth and then automatically connect to WIFI. But when I connect it to car then it doesn't connect to WIFI because of which android auto doesn't start in the car. 

What am I doing wrong?

manio, 2024-11-30:
What connection strategy are you using (default?). I described the whole connection process [here](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical).
Maybe you find it useful to see whats going wrong (and when).

tanmay380, 2024-11-30:
1) I plug the pi to the car unit

2) phone is connected to cars bluetooth and raspberry pi via bluetooth

Do I need to do some other steps too? As I read your tutorial i think I'm
following all of them.

But now wifi doesn't connect and bluetooth doesn't get disconnected which
was happening when I was using pi connected to a normal power source.

tanmay380, 2024-11-30:
Also on my car unit AA option isn't getting enabled after connecting pi.
Will it get enabled after all the connections are successful.

manio, 2024-11-30:
> Also on my car unit AA option isn't getting enabled after connecting pi.
> Will it get enabled after all the connections are successful.

If you are using default connection strategy (I don't know because you don't answer my question) this is normal. Pi is starting connecting to the car *After* your phone connects first.

tanmay380, 2024-11-30:
yes i'm using default connection strategy.

tanmay380, 2024-11-30:
i'll try to change the usb cable and will try again.

tanmay380, 2024-11-30:
At home.mp4
<https://drive.google.com/file/d/1UmXKBoB_fb_rn3HXvJVMJxjWpi53edip/view?usp=drive_web>
 Car.mp4
<https://drive.google.com/file/d/1MjnXGhNyfQWpa_gCzAwzAB4ecxTNRgp7/view?usp=drive_web>
Changed the cable and still the same result. I'm uploading the video.

manio, 2024-11-30:
What Pi you're using? I thinks that a logs would be very useful...

tanmay380, 2024-11-30:
Pi zero 2 W.

I'll share the logs by tomorrow,