nisargjhaveri/WirelessAndroidAutoDongle #231: unable to open MTP device / couldn't find matching udev device

KreAch3R, 2024-12-02:
Setup:

Headunit: Raspberry Pi 4: Raspbian 10 buster
Dongle: Raspberry Pi Zero 2W: latest release
Android Auto headunit: OpenAuto-Pro (A fork of Crankshaft).

Issue:
The dongle works. When Android Auto is connected and the image is projected successfully to the Headunit RPI4, an error popup shows up with:
`unable to open MTP device "001,012"`
or
`couldn't find matching udev device`

It happens every time. It doesn't affect the functionality.

I suspect it is because of this: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/pull/3 and the MTP implementation somehow is conflicting with Raspbian. Maybe because it expects a rw device and gets a ro? 

@BluemediaDev  @nisargjhaveri I hope I get your views on this on how to fix it / bypass it.

manio, 2024-12-03:
Maybe it is because switching dongle modes (accessory <-> default) or switching too fast. You can see if it is working without it ([aa-proxy-rs](https://github.com/manio/aa-proxy-rs) doesn't have that switching by default but you can enable it with a "legacy" command line switch if it doesn't work.

KreAch3R, 2024-12-03:
Thanks @manio I will give aa-proxy-rs a try, by manually installing it onto the image and get back to you. Can you check my other issue #226?

KreAch3R, 2024-12-03:
Well I gave your image a try, it does appear that the MTP error is not thrown, but the connection doesn't stay on, it goes into a loop of connecting for a couple of seconds, then dropping the connection, then successfully reconnecting etc. It's not usable as is, I'll go back for the moment. Any suggestions?

manio, 2024-12-04:
> Any suggestions?

Hard to tell anything without a logs.

nisargjhaveri, 2024-12-05:
> unable to open MTP device "001,012"

Our MTP implementation is very very basic. We don't allow opening or listing anything from MTP right now, it is just to provide basic communication for the headunit to think it is corrected to a phone. Maybe it is trying to open via mtp and failing? You can play around with MTP config and see if it makes any difference.

For the switching, many headunits seems to expect MTP device first. While accessory mode should directly work always based on documentation, doesn't seem like headunit implementations consistently support this. So presenting as MTP and then switching seemed to give wider compatibility by default.

KreAch3R, 2024-12-05:
> Our MTP implementation is very very basic. We don't allow opening or listing anything from MTP right now, it is just to provide basic communication for the headunit to think it is corrected to a [...]

Ok, I'll try to play around and check the logs and get back to you. I just wanted to know if it's a known bug or misconfiguration. 

Regarding the switching, I wouldn't want to mess with that. I played around with the connection strategy and I had dropped connections. The "phone first" method works pretty well. The only problem is the timing, because the headunit rpi does take a 30 second boot up time while the zero2w dongle boots up in seconds and has already connected to the phone twice before the headunit is up. 

I searched around in the source but didn't find any other delay parameter I could change besides the `boot_delay` parameter, to slow down the dongle firing. Is that the recommended way?

KreAch3R, 2024-12-05:
Ok, so [boot_delay](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/raspberrypi/config.txt) is the correct parameter to mess around. Setting it at around `25`, as long as my headunit RPI takes to boot, makes it connect around the same time the headunit is ready to accept the USB connection.

Regarding the MTP error, I fixed it for my case. I needed to override the default `69-libmtp.rules` by copying the `/lib/udev/rules.d/69-libmtp.rules` into `/etc/udev/rules.d/` of **_the Headunit RPI_** and exclude the dongle from libmtp:
```
...
# Some sensitive devices we surely don't wanna probe
# AA Wireless Dongle
ATTR{idVendor}=="18d1", ATTR{idProduct}=="4ee1", GOTO="libmtp_rules_end"
...
# Color instruments
ATTR{idVendor}=="0670", GOTO="libmtp_rules_end"
...
```

I used the values from here: [umtprd.conf ](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/umtprd/umtprd.conf) and this is a working `journalctl -fe`:

```
Dec 06 00:30:14 raspberrypi mtp-probe[1849]: bus: 1, device: 10 was not an MTP device
Dec 06 00:30:41 raspberrypi kernel: usb 1-1.4: USB disconnect, device number 10
Dec 06 00:31:15 raspberrypi kernel: usb 1-1.4: new full-speed USB device number 11 using xhci_hcd
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: new high-speed USB device number 12 using xhci_hcd
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: New USB device found, idVendor=18d1, idProduct=4ee1, bcdDevice= 6.01
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: Product: AA Wireless Dongle
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: Manufacturer: My Own
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: SerialNumber: 0123456
Dec 06 00:31:27 raspberrypi kernel: usb 1-1.4: USB disconnect, device number 12
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: new high-speed USB device number 13 using xhci_hcd
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: New USB device found, idVendor=18d1, idProduct=2d00, bcdDevice= 6.01
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: Product: AA Wireless Dongle
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: Manufacturer: My Own
Dec 06 00:31:28 raspberrypi kernel: usb 1-1.4: SerialNumber: 0123456
Dec 06 00:31:28 raspberrypi mtp-probe[1939]: checking bus 1, device 13: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4"
Dec 06 00:31:28 raspberrypi mtp-probe[1939]: bus: 1, device: 13 was not an MTP device
Dec 06 00:31:28 raspberrypi mtp-probe[1940]: checking bus 1, device 13: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4"
```

The real issue is described [here](https://sourceforge.net/p/libmtp/bugs/1897/) and [here](https://superuser.com/a/1170917/414681). Basically, a lot of MTP devices are connecting and disconnecting fast enough in the beginning that `libmtp` throws a `PANIC` and that goes through `gfvs` to the userspace and presents the error. 

```
PTP_ERROR_IO: failed to open session, trying again after resetting USB interface
LIBMTP libusb: Attempt to reset device
LIBMTP PANIC: failed to open session on second attempt
```

It is so common that I am marking it as a libmtp bug in my mind, but in this case, you can stop `mtp-probe` from trying to probe AAWirelessDongle without messing the AA wireless connection.

So, I'm considering this issue as closed. I only have one question towards the original authors of the MTP implementation. Should this value be `rw`: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/4fa8c338fca608c0e7b074cd37ea863b57ae5288/aa_wireless_dongle/board/common/rootfs_overlay/etc/umtprd/umtprd.conf#L3

given that the whole image is mounted as `ro`? I changed it to `ro` as well, didn't notice any new problems, but I don't know what problems to expect.

KreAch3R, 2024-12-22:
I've been successfully using `umtprd.conf`'s `ro, locked` value for a couple of days now using `aa-proxy-rs` and it behaves normally. 

`aa-proxy-rs` doesn't switch between the `default` and `accessory` modes without setting the `--legacy` argument, so this patch is unnecessary there. 

I consider this issue solved, so I'll close it.

closed 2024-12-22 by KreAch3R