nisargjhaveri/WirelessAndroidAutoDongle #232: Has anyone tried to get this working on ESP32 boards with WiFi/BT?

Artasdmc, 2024-12-04:
I've got spare ESP32's left over and I thought this would be a great use for one of them. Haven't seen any posts regarding these boards.

nisargjhaveri, 2024-12-05:
We ideally need a usb 2.0 support, and a good wifi connection. While 2.4GHz wifi works, ideal would be a 5GHz wifi support. I believe esp32 doesn't meet any of these, it might be a uphill task to get this working well. Also, we'd need a completely different architecture to make it work on esp32, the current Linux based one might not work due to resource constraints.

manio, 2024-12-05:
This project is linux-based. ESP32 boards are for different purposes - you are writing the whole program which is running on them from scratch - or you are using prepared Espressif realtime OS.

In other words: this project is using Linux - which is a ready operating system, and a whole related tools (bluetooth bluez stack, kernel drivers etc) - so we are using existing "blocks" which are easy to use.

If you want to use ESP32 then you have to dig much more low-level - and probably write everything from scratch: bluetooth handling, USB OTG handling (if the ESP is hardware-capable of this), WiFi and last but not least: the whole AA proxy which is doing the on top of those. I am not saying that it is impossible (it would be even interesting ;), but even if you use a libraries for this (which is provided), it is still a huge amount of work, because you need to write just *everything* to has it working. The side effect is the speed though - it would probably start and work faster, but this is just a guess - because a CPU power is also relevant here (but there are also range of ESP's out there based on extensa and risc architectures).