nisargjhaveri/WirelessAndroidAutoDongle #233: Can't connect to MA:CA:DD:RE:SS here.

yuppicide, 2024-12-13:
I had something happen the past 2 days out of nowhere and I don't know why. It was working then out of nowhere it wasn't working and every 2 seconds my phone popped up a message Can't Connect to and had what I assume is the RPI4's mac address listed. It went away and came back today. I don't know what actually fixed it. I did say "Forget" to the AAWirelessDongle and then tried again, I disconnected from my car and reconnected. Eventually it started working again.

yuppicide, 2024-12-13:
I had to disconnect the device. I can't get it to work and every 2 seconds my audio cuts out and it says connect a device when it's safe to do so or something. 

If I connect my phone to my car I get audio and phone, but I can't get the AAWirelessDongle to work anymore and I use it for maps.

manio, 2024-12-13:
Are you describing something that happened in the past (that's the impression after initial post) or the problem is still there currently?

yuppicide, 2024-12-13:
I wrote it in three original post it's happening the past 2 days.. meaning yesterday and today. The issue where it can't connect came up once previously but then started working. This time it's been like 2.5 days so far I can't figure out how to get working.

Currently I got forget car and forget AAWirelessDongle and then I re paired them but no luck.

manio, 2024-12-14:
I hope Nisarg would be OK with this request:
Can you also try this with my [project](https://github.com/manio/aa-proxy-rs) and provide a logs? The logs are the most relevant part because we can visually see what is going on...

yuppicide, 2024-12-14:
> I hope Nisarg would be OK with this request: Can you also try this with my [project](https://github.com/manio/aa-proxy-rs) and provide a logs? The logs are the most relevant part because we can visually see what is going on...

I will give your fork a try soon as I can.