nisargjhaveri/WirelessAndroidAutoDongle #234: How to build the aagw source code without buildroot?

aggo15, 2024-12-17:
This is not an issue, sort of like a request to the repo owner. 

I've tried the buildroot image on a Pi0W and it work just fine. However, I'd like to add in some existing scripts from my side that works with GPIO. I prefer to run the AA wireless dongle code on an official Raspberry Pi OS instead of buildroot image. Naturally I tried to read your code and attempt to build the aagw source independently from buildroot but my meager C++/Make knowledge failed me.

Do you mind to provide a build script that can run on official Raspberry Pi OS? Thank you in advance.

manio, 2024-12-17:
If you prefer to run it on the RaspberryPI OS then I can tell you that it's not so easy, eg: you need to build a customized kernel modules for OTG/gadget and do configuration of those. I did similar thing (using GPIO and serial) with [my LoRa project](https://github.com/manio/lora-gw-rs), but despite using [aa-proxy-rs](https://github.com/manio/aa-proxy-rs) instead of `aawgd` I am still using the whole buildroot from this project as it is much easier (all hard work done by Nisarg).

Anyway: you can just build it as any other regular application. Full image installation steps are eg here:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/BUILDING.md#build-with-manual-setup
and if you want a aawgd alone, just go to:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/tree/main/aa_wireless_dongle/package/aawg/src
and BR build script is here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/package/aawg/aawg.mk

aggo15, 2024-12-17:
@manio thanks for the pointer. I'll have a look. Wonder if anyone attempt to build this as Docker image and run as container? It will be much more easier to run on official Raspberry Pi OS if it exists.

manio, 2024-12-17:
> It will be much more easier to run on official Raspberry Pi OS if it exists.

First of all you have to communicate somehow from the container to the "outside" world... it's not possible to run a custom kernel module in docker - you need to have it in your own Pi OS. Look how it is [working](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical) first.

aggo15, 2025-01-04:
Thank you for the information. Let me try to play around a little with the codes. I only managed to build the buildroot image so far and still working out to get only the bits I need. Will close this issue.

closed 2025-01-04 by aggo15