nisargjhaveri/WirelessAndroidAutoDongle #235: Android Auto error 14

Ikif, 2024-12-18:
Several days before I started to get this error (14).
Google Play Services on my phone is updated (Samsung Note 8) .
Time and date on my phone and on the car media is correct.
One thing I can assume , maybe time/date on the RPzero is not correct...  the SD card is on "system read-only" state.

Please advise.

manio, 2024-12-18:
Time/date on the RPI doesn't matter at all. The aawgd daemon is forwarding data transparently from car to phone. R/O of SD card also doesn't matter. I had the "RED" android error only when the car headunit has no time set (as soon as it is set from GPS it is then working fine) and also when I was playing with different AndroidAuto application on the phone: eg if I had very old version than I also had this error.

First I'd check direct USB cable connection...

Ikif, 2024-12-18:
The AA version is not latest : ver. 12.3.642624

manio, 2024-12-18:
I have currently have `13.3.644654-release` but I think it is dependent on different factors (i.e. Android version).

Is it working when you connect your phone to the headunit directly with a USB cable?