nisargjhaveri/WirelessAndroidAutoDongle #241: Pi Zero W Force Connect to My Phone Even though Bluetooth, Location and Wifi Is Off

aggo15, 2025-01-04:
So far the Pi Zero W as wireless AA dongle is working great. However, there is this strange behavior I observed when using the dongle. 

There are times where I will turn off bluetooth and wifi to disconnect my phone from the dongle before I get off my car. Occasionally, the dongle will reconnect to my phone again if I stay close to my car even though I turned off wifi and bluetooth on my phone.

My first question is, is this behavior normal? It shouldn't be able to connect to my phone if I explicitly turn off everything on my phone right? Somehow, something in the dongle managed to turn on wifi and bluetooth on my phone without my permission. That don't really sit well with me.

Second question is, how do I properly tell the dongle to disconnect from my phone until I request to connect to the dongle via bluetooth? Is there any steps I could follow in order to tell the dongle don't connect again unless I explicitly request to?

Thanks in advance.

manio, 2025-01-04:
> My first question is, is this behavior normal? It shouldn't be able to connect to my phone if I explicitly turn off everything on my phone right? Somehow, something in the dongle managed to turn on wifi and bluetooth on my phone without my permission. That don't really sit well with me.

This is rather impossible - if you disconnect both (Wifi and BT) then how it could connect? :)
I only noticed that: when AA is working, and you disconnect WiFi, then it is trying to automatically reconnect and the android will bring WiFi on (it may be also other way around IIRC). Anyway: here is how I am doing this:
I am clicking "Disconnect Android Auto" from the phone and then very fast disable bluetooth and wifi. After enabling only wifi I can normally use phone without android auto.

> Second question is, how do I properly tell the dongle to disconnect from my phone until I request to connect to the dongle via bluetooth? Is there any steps I could follow in order to tell the dongle don't connect again unless I explicitly request to?

This all depends on connection strategy you are using. Look here what options you have:
https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/ee88171a9a4e2adc9a065a1608ebaa6266efb354/aa_wireless_dongle/board/common/rootfs_overlay/etc/aawgd.env#L1
In case you're using my daemon you can remove "-c" [parameter](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#usage).

Can I also ask the desired behavior you like? Why are you disconnecting AA yourself? In my case I am only doing it in one specific case - when I am charging my electric vehicle in the garage - because then my RaspberryPi is up and running and still trying to maintain the connection with my phone, but otherwise it will just shutdown.

nisargjhaveri, 2025-01-04:
By default, we keep trying to connect to phone via Bluetooth. So, if you're just disconnecting, it'll automatically try again.

If the bluetooth is disabled in the phone, it should not connect. If it is still connecting, I'd suspect phone is misbehaving and not disabling bluetooth properly.

Disabling wifi is not enough. As @manio mentioned, one AA connects via bluetooth, it can automatically turn wifi on and connect via wifi as well in many phones.

aggo15, 2025-01-07:
> This is rather impossible - if you disconnect both (Wifi and BT) then how it could connect? :) I only noticed that: when AA is working, and you disconnect WiFi, then it is trying to automatically [...]

Yes I think after tapping the "Disconnect Android Auto" option on my phone and kill my phone's BT/Wifi next, work the best without the dongle reconnecting to my phone. I've tried it on my car for a few days now I did not see the reconnect happens again. Perhaps disconnecting the BT instead of using "Disconnect Android Auto" causing it to misbehave.

> Can I also ask the desired behavior you like? Why are you disconnecting AA yourself? In my case I am only doing it in one specific case - when I am charging my electric vehicle in the garage - [...]

Similar use case as you. When I try to charge my electric vehicle, usually I'll get out of my car after disconnecting AA on my phone and get the CC2 plug to my car. When I turned on the charger app on my phone to control the charger, then somehow AA reconnected to my phone and causing my screen to blackout for a seconds. It is not serious problem, just a mild inconvenience as I have to disconnect the AA again. 

Ideally, if the AA dongle received a disconnect request/event from the phone, it should stop attempting to reconnect afterwards until it received a connection request again from the phone.

aggo15, 2025-01-07:
> By default, we keep trying to connect to phone via Bluetooth. So, if you're just disconnecting, it'll automatically try again.
>
> If the bluetooth is disabled in the phone, it should not connect. If [...]

I see. For now I will follow @manio recommendation to use the "Disconnect Android Auto" option and the issue did not appear again. Will close the issue for now.

closed 2025-01-07 by aggo15