nisargjhaveri/WirelessAndroidAutoDongle #243: Wi-Fi Direct Support

KadenK, 2025-01-16:
The raspberry pi, and a majority of other boards should be capable of connecting to a phone through Wifi Direct rather than posing as an access point. Normally, this is how a built-in wireless android auto system works which allows the phone's wifi connection to remain open to a internet-connected network. 

An additional benefit of Wifi Direct capability would be to reduce the problem that many of us have of the car leaving the power supply on, and therefore the Pi powered, for around 10 minutes after the car is turned off. With WiFi Direct, this won't be as much of an annoyance.

To those who know a bit more about this project, is this reasonable? If so, I would be glad to assist in the development, but it would be helpful if I was given a place to start.

manio, 2025-01-17:
Hi!
> The raspberry pi, and a majority of other boards should be capable of connecting to a phone through Wifi Direct rather than posing as an access point. Normally, this is how a built-in wireless [...]

Oh... this seems interesting... so in this mode other AA-wireless dongles are working like eg [AAWirelessTwo](https://www.aawireless.io/en/products) and similar?

Two things comes to my mind:
1. even if the phone is WifiDirect-connected, is it able to freely use the same WiFi adapter for other purposes? I doubt, especially if the other APs are on different channels then I think it would harm either our AA transfer or the other internet-enabled AP (but I could be wrong, please confirm if I am).
2. if we set the Raspberry in WifiDirect, we should also make the "old" way of connection available - eg for connecting remotely for debuging purposes.

> An additional benefit of Wifi Direct capability would be to reduce the problem that many of us have of the car leaving the power supply on, and therefore the Pi powered, for around 10 minutes after the car is turned off. With WiFi Direct, this won't be as much of an annoyance.

Not quite - maybe this is more car-dependent thing. In my car it stays on for several minutes - true, but the problem is that from the dongle point of view - we are not capable of telling if it should work or not - the system is not passing any information that the car has stopped, driver is out, etc. It is just working because the car media system still getting power, is working and eg. still playing music from phone. This also applies more when the infotainment is on eg during charging the car. But I would not consider this a something wrong - even more - I am sometimes doing something in garage and keeping my phone connected for purpose.

nisargjhaveri, 2025-01-19:
Do you have more info on how Wifi direct would work? The last I had checked and understood, the initiation protocol via bluetooth was expecting a traditional AP, at least that is what I understood.

KadenK, 2025-01-26:
Sorry it's taken me a bit to respond, I got really busy.

> even if the phone is WifiDirect-connected, is it able to freely use the same WiFi adapter for other purposes?
Yes, when connected using WiFi Direct, there should not be any interruptions to regular WiFi usage.  Many android phones support around 5 WiFi Direct connections and a WiFi connection at once. You can think of WiFi Direct as similar to Bluetooth. Often screen mirroring, wireless printers, and file transfers use WiFi Direct.

> if we set the Raspberry in WifiDirect, we should also make the "old" way of connection available - eg for connecting remotely for debuging purposes.
I fully agree on this, although once it gets stable enough, I would suggest making this the default. 

As far as the part about resolving some of the issues of the dongle not turning off because of the car, I say this because my main concern about it is that if I stay within WiFi range of my car, my phone will not disconnect and connect to my home WiFi, resulting in weird behavior in apps and network speed. If I do disconnect from the dongle WiFi and I am still within Bluetooth range, the dongle will make the BT connection and force me onto the dongle's WiFi network. WiFi Direct will solve this because you can separately connect to a internet-connected AP as well.

> Do you have more info on how Wifi direct would work? 
Are you asking about how it would specifically be implemented in this project? Not really. I'm familiar with Raspbian WiFi configurations, but not so much embedded linux systems, although I'm happy to help research. This ticket is mostly to propose an idea and gather info about if it seems possible from those who would probably know better before making a real commitment to the feature.

> The last I had checked and understood, the initiation protocol via bluetooth was expecting a traditional AP, at least that is what I understood.
Again, I don't know too much about the specific communication protocol Android Auto expects, as far as I'm aware there isn't really any documentation and I don't have the skills to monitor and discover the connection protocols.

manio, 2025-01-27:
> my phone will not disconnect and connect to my home WiFi, resulting in weird behavior in apps and network speed

Thanks for clarification, this is now clear to me - as the missing puzzle was that I didn't know that the phone can use both WiFi connections at a time :)

Regarding the WiFi Direct itself: I can see that from the linux point of view there is eg [this thread](https://stackoverflow.com/questions/31877144/how-to-set-up-a-wifi-direct-connection-between-android-and-linux) but I didn't dig deeper. If I can propose something: the [iwd](https://wiki.archlinux.org/title/Iwd) seems to be a good candidate because:
_"The core goal of the project is to optimize resource utilization by not depending on any external libraries and instead utilizing features provided by the Linux Kernel to the maximum extent possible."_
so it seems suitable for this kind of embedded buildroot environment, but I don't know how and if it supports WiFi Direct...

Anyway - this topic is really interesting :)

gamelaster, 2025-01-27:
Last time I played with WiFi Direct on Linux (and generally), it was quite pain, it was not reliable. I remember that even authors of AAWireless in Headunit Reloaded project, they abandoned WiFi Direct because it did not worked well.
For WiFi Direct support, it needed external application on phone to do the handshake, connection and then send intent to AA to connect to the server.

From the Android Auto manuals from 2018, they did not planned WiFi Direct support. This maybe changed, but unless someone who have access to latest documents or reverse engineered the AA apk, we do not know. I don't think there is simple seamless integration in terms of WiFi Direct. And I also don't know if there are any official AA supporting headunits, which uses WiFi Direct, it's always AP.
That means adding WiFi Direct support might not be easy to do, and will require the additional app in phone for auto-connect.

manio, 2025-01-27:
@gamelaster Thank you for this valuable input on this. Installing additional soft on phone is indeed a pain and not worth the benefits IMO.

I am only wondering if there are any closed-source boxes, eg [AAWireless TWO](https://www.aawireless.io/en/products/aawireless-two), etc. which are using this, or they simply use AP mode?

gamelaster, 2025-01-27:
@manio they very likely use AP mode, or if WiFi Direct, then companion app is needed (unless it's really AA feature, which I am not aware of).