nisargjhaveri/WirelessAndroidAutoDongle #245: Changing WiFi Channel?

uvmatt3000, 2025-02-28:
Apologies if this has been asked before, however incannot find anything related to my exact issue.

I am using a Pi 3A+ and everything is working absolutely flawlessly _except_ when I drive through certain areas and the music begins to stutter.. returning to normal.

When I go to skip a track, there is then a one second or so delay from when I press the button to skip track to when it actually actions it.

Rebooting the stereo (which reboots the Pi) fixes this until the next time I drive through a certain area.

I wondered if it was interference on the same wireless frequency, so I set up the app 'WiFi Analyzer' on my phone.

Sure enough, the exact area I drive through where the music cuts out, WiFi channel 36 is crammed with SSIDs.

I thought i would try to be clever and manually change the WiFi channel by modifying the **hostage.conf** file line **channel=36** with a number of my choosing (within the 5Ghz channels) however, when I change this number, the Pi then doesn't give out any WiFi at all.

Changing it back to channel 36 works fine again.

My question is, can I change the WiFi channel to one that will avoid channel 36 interference from other access points?

Channel 64 or 100 would be perfect, but it breaks the Pi giving out WiFi.

Could anyone advise on this?

Help would be much appreciated, thank you!

manio, 2025-03-01:
> however, when I change this number, the Pi then doesn't give out any WiFi at all.

This is weird. Are you sure you did it properly?

uvmatt3000, 2025-03-01:
> > however, when I change this number, the Pi then doesn't give out any WiFi at all.
> 
> This is weird. Are you sure you did it properly?

Thank you manio for the fast reply!

Yes it's very strange, even if I change the country code to one that fully supports that channel, it just doesn't work.

I've tried multiple different ways to edit that file, to no avail, even using different editing methods to put 36 back onto that line, 36 always gives out wifi after booting the Pi. 

I also did this via SSH directly to the Pi whilst it was on channel 36 and changed it on the fly, but this did not help.

If I then remove the SD card, insert into Linux PC and browse to the **hostapd.conf** file and change the number back to 36, boot up the Pi, sure enough the **AAWirelessDongle** AP shows again.

Annoyingly, I can't see any logs afterward I change it because the 3A+ does not have an ethernet socket to connect to and logs are not persistent.

Even the CYW43455 chipset datasheet clearly states that the frequencies I want to use are supported.

I wonder if I configure the journal to save to disk it might give more clues?

It is a shame my issue is so specific otherwise you would be able to recreate the issue for your device maybe?

Thank you for looking at my issue though, I'm sure it will turn out to be something simple, this may just be a case of trial and error to see what happens...

manio, 2025-03-01:
> Annoyingly, I can't see any logs afterward I change it because the 3A+ does not have an ethernet socket to connect to and logs are not persistent.

If you have a free USB port then you can use an USB-ethernet adapter (tested). There is also a way to connect it directly to laptop in some special mode - a simple USB cable is enough IIRC (didn't test it though).

Are you sure you also set `hw_mode` to `a` ?
This is a sample snippet for 5GHz:
```
## 5G
hw_mode=a
channel=36
country_code=US
ieee80211d=1
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
```
I've got it from: https://raspberrypi.stackexchange.com/questions/91074/how-to-configure-hostapd-for-a-5ghz-network

> It is a shame my issue is so specific otherwise you would be able to recreate the issue for your device maybe?

I don't have any Pi with 5GHz unfortunately :(

uvmatt3000, 2025-03-01:
Hi manio,

Thank you for the continued assistance, I think I have figured it out...

I believe this was simply down to the combination of country code and what channel I was testing with.

From your first message you posted, I was confident I was on the right track so I have been doing more research into this.

Doing some Googling, I came across this webpage:

https://forums.raspberrypi.com/viewtopic.php?t=335268

Sure enough, after reading the website and setting the channel to 48, then booting the Pi, I got this:

[image]

Running `iw list` on the PI showed this:

[image]

This made me think that the Pi 3A+ is unable to use most of these frequencies?

After some more Googling, I came across this post:

https://github.com/raspberrypi/firmware/issues/1246#issuecomment-779897102

So I then used country code US and set the channel to 64 and lo and behold:

[image]

Case closed! Now looking forward to a road test tomorrow to see if it works, or if I've inadvertently chosen channel that has even more interference than 36....

manio, apologies for wasting your time and thank you for bearing with me, it's much appreciated! :)

closed 2025-03-01 by uvmatt3000

manio, 2025-03-02:
> manio, apologies for wasting your time and thank you for bearing with me, it's much appreciated! :)

Not a problem at all, I'm glad that it leads you to a solution! And I am also learning from this, btw :)