nisargjhaveri/WirelessAndroidAutoDongle #246: Keep disconnecting on Android 13 with pi zero w

blaze0004, 2025-03-08:
As everyone is keep posting about this great project, I though I would also try it using my raspberry pi zero w. I installed latest pre-build and tried to connect as mentioned but it keep disconnecting after every few seconds. I would really appreciate if someone can help on this

It seems the Bluetooth is connected and WIFI info is also shared but once's the WIFI is connected it keep disconnecting and showing notification android auto available trying to connect.

I tried with my Google DHU Emulator it just keep waiting for the device on that and on the car unit it shows device not connected error.

Please are some logs, see if someone can help on this:
`
Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2024-08-03 03:46:38 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61 (runner@fv-az1490-423) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 Sat Aug  3 04:19:00 UTC 2024
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writeback
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:50:B5:7D vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 340196K/421888K available (7906K kernel code, 702K rwdata, 2184K rodata, 256K init, 498K bss, 16156K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000057] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000158] bcm2835: system timer (irq = 27)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000943] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001440] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.001518] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060291] CPU: Testing write buffer coherency: ok
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060397] pid_max: default: 32768 minimum: 301
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060587] LSM: Security Framework initializing
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060798] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.060868] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.062267] cgroup: Disabling memory control group subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064154] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064236] cblist_init_generic: Setting shift to 0 and lim to 1.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.064556] Setting up static identity map for 0x8220 - 0x8258
Jan  1 00:00:03 buildroot kern.info kernel: [    0.065813] devtmpfs: initialized
Jan  1 00:00:03 buildroot kern.info kernel: [    0.076646] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Jan  1 00:00:03 buildroot kern.info kernel: [    0.077046] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.077158] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.087595] pinctrl core: initialized pinctrl subsystem
Jan  1 00:00:03 buildroot kern.info kernel: [    0.089292] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Jan  1 00:00:03 buildroot kern.info kernel: [    0.091787] DMA: preallocated 1024 KiB pool for atomic coherent allocations
Jan  1 00:00:03 buildroot kern.info kernel: [    0.097571] audit: initializing netlink subsys (disabled)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.099021] thermal_sys: Registered thermal governor 'step_wise'
Jan  1 00:00:03 buildroot kern.info kernel: [    0.099333] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.099423] hw-breakpoint: maximum watchpoint size is 4 bytes.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.099615] Serial: AMBA PL011 UART driver
Jan  1 00:00:03 buildroot kern.info kernel: [    0.106967] bcm2835-mbox 2000b880.mailbox: mailbox enabled
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.110676] audit: type=2000 audit(0.090:1): state=initialized audit_enabled=0 res=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.130723] raspberrypi-firmware soc:firmware: Attached to firmware from 2023-10-17T15:42:39, variant start
Jan  1 00:00:03 buildroot kern.info kernel: [    0.140749] raspberrypi-firmware soc:firmware: Firmware hash is 30f0c5e4d076da3ab4f341d88e7d505760b93ad7
Jan  1 00:00:03 buildroot kern.info kernel: [    0.187430] kprobes: kprobe jump-optimization is enabled. All kprobes are optimized if possible.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.191003] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[... 271 lines cut ...]
Jan  1 00:01:48 buildroot user.info aawgd[217]: Path: /org/bluez/hci0/dev_C8_58_95_A0_67_FC, fd: 15
Jan  1 00:01:48 buildroot user.info aawgd[217]: Bluetooth connected to the device
Jan  1 00:01:48 buildroot user.info aawgd[217]: AA Wireless NewConnection
Jan  1 00:01:48 buildroot user.info aawgd[217]: Path: /org/bluez/hci0/dev_C8_58_95_A0_67_FC, fd: 16
Jan  1 00:01:48 buildroot user.info aawgd[217]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
[... cut ...]
Jan  1 00:01:50 buildroot kern.info kernel: [  110.645628] dwc2 20980000.usb: new device is high-speed
Jan  1 00:01:50 buildroot user.info aawgd[217]: Bluetooth adapter was powered off
Jan  1 00:01:50 buildroot kern.info kernel: [  110.679076] android_work: sent uevent USB_STATE=CONNECTED
Jan  1 00:01:50 buildroot kern.info kernel: [  110.681492] dwc2 20980000.usb: new address 44
Jan  1 00:01:50 buildroot kern.info kernel: [  110.823806] android_work: sent uevent USB_STATE=CONFIGURED`

Thanks to @nisargjhaveri and other contributors.

nisargjhaveri, 2025-03-08:
> Jan  1 00:01:03 buildroot user.info aawgd[217]: USB Manager: Timeout waiting for accessory start request

Looks like the headunit never connects. If you're facing this with DHU as well, my best guess would be a faulty USB cable. Can you make sure the cable you're using is a working data cable?

blaze0004, 2025-03-08:
I am using type c to type c data cable on phone as well as usb a to c and micro on rpi.

These are not normal charging cables as I am able to transfer data at high speed with these as well.

Are there any steps I can follow to check if the data cable is faulty or to properly connect with dhu both phone and rpi.

My DHU keeps looking for the rpi as well as my phone in accessories mode. I think I tried with adb earlier with my phone that worked but no result with accessories mode.

Thanks

manio, 2025-03-09:
> Also is there any documentation on how this whole wireless android auto works under the hood.

If you're talking about initiating connection: [here](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical) you can find something. Regarding protocol, you can look [here](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#similarother-open-source-androidauto-related-projects).

blaze0004, 2025-03-09:
> If you're talking about initiating connection: [here](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical) you can find something. Regarding protocol, you can look [...]

Thanks, will look into that. ♥️

blaze0004, 2025-03-09:
Please correct me, So I was reading this kernel development stuff and trying to understand what we could be doing in this project.

So basically we have to build a minimalist distribution and write a driver read and write data to the head unit.

We are creating a tcp connection over the local wifi net and reading and writing the data sent over the tcp connection to the device. and by device I mean char dev.

@nisargjhaveri / @manio is this what we are doing here?

manio, 2025-03-10:
@blaze0004 Sorry, I don't understand. What is your question exactly and what you want to achieve?

blaze0004, 2025-03-10:
> @blaze0004 Sorry, I don't understand. What is your question exactly and what you want to achieve?

1. I was just trying to understand how this works, so I was just asking if you can confirm my understanding.
2. I was asking how I can identify the right cable, I am using a data cable with high speed data transfer rightnow. I even tried with my original type c to type c cable to connect my phone with dhu directly.

manio, 2025-03-10:
> I was just trying to understand how this works, so just asking if you can confirm my understanding.

This I think was in the links I provided?

> I was asking how I can identify the right cable, I am using a data cable with high speed data transfer rightnow. I even tried with my original type c to type c cable to connect my phone with dhu directly.

Following Nisarg advice: a good cable is important. And even if cable is working (have a data pins) doesn't automatically mean it is **good**. Good cable have a good electric parameters - generally cheap cables tends to be crap.

As Nisarg told you, the main problem you've got is:
```Timeout waiting for accessory start request```. This **could** mean some problems with cable. You can also try [my proxy](https://github.com/manio/aa-proxy-rs) if you like - this "waiting for accessory" functionality is disabled by default, as not all units needs this thing, eg my car headunit does works well without this, but I **am aware** there are units which doesn't work without it (to name some: Sony XAV-AX5500). But if you need compatibility you can enable it with a switch.

blaze0004, 2025-03-10:
I'm now using different cable which I have been using in my car with my phone directly for android auto. It is also not working with the DHU, it keep showing no device found errors.

Will check if the same is occurring with your proxy

Thanks @manio

closed 2025-04-13 by blaze0004