nisargjhaveri/WirelessAndroidAutoDongle #247: Segmentation Fault on aawg daemon

RicardoCarvalho01, 2025-03-11:
## Description  
When calling `DBus GetManagedObjects`, the `aawgd` daemon results in a segmentation fault. This occurs consistently across multiple builds from source.  

## Steps to Reproduce  
1. Build and install `aawgd` from source for Raspberry Pi Zero 2 W.  
2. Normal boot or manually start the `aawgd` daemon.
3. Observe the segmentation fault when invoking `GetManagedObjects` (print info or debug).

## Expected Behavior  
The program should return the expected D-Bus managed objects without crashing and the daemon should run normally.

## Actual Behavior  
A segmentation fault occurs when calling getManagedObjects():  
```cpp
DBus::ManagedObjects BluetoothHandler::getBluezObjects() {
    std::shared_ptr<DBus::ObjectProxy> m_bluezRootObject = m_connection->create_object_proxy(BLUEZ_BUS_NAME, BLUEZ_ROOT_OBJECT_PATH);
    DBus::MethodProxy getManagedObjects = *(m_bluezRootObject->create_method<DBus::ManagedObjects(void)>("org.freedesktop.DBus.ObjectManager", "GetManagedObjects"));

    return getManagedObjects();
}
```

```
# System Logs (/var/log/messages)
Jan  1 00:00:04 AAutoPi_Wireless user.info console: Starting aawgd: OK
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: AAutoPi Wireless
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: Starting uevent monitoring
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: Uevent monitoring started
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: Initializing USB Manager
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: USB Manager: Disabled all USB gadgets
Jan  1 00:00:04 AAutoPi_Wireless user.info aawgd[275]: USB Manager: Found UDC 3f980000.usb
...

# Daemon Execution Output (aawgd)
aawgd[362]: AAutoPi Wireless
aawgd[362]: Starting uevent monitoring
aawgd[362]: Uevent monitoring started
aawgd[362]: Initializing USB Manager
aawgd[362]: USB Manager: Disabled all USB gadgets
aawgd[362]: USB Manager: Found UDC 3f980000.usb
Segmentation fault
```

## Environment
- Hardware: Raspberry Pi Zero 2 W
- Build Method: Built from source

## Additional Notes
This issue happens on every build from source.
Could be related to D-Bus or how GetManagedObjects is invoked.

manio, 2025-03-11:
This is probably a duplicate of #196 (and also a reason I rewrited it in Rust ;)

nisargjhaveri, 2025-03-14:
@RicardoCarvalho01 does this happen on pre-built versions as well?

@manio, the other bug you mentioned fails much later than what is being described here?

manio, 2025-03-14:
> the other bug you mentioned fails much later than what is being described here?

I don't remember ... but IIRC in the #196 it was always when starting BT (and the bluetooth was busy/not ready), here in @RicardoCarvalho01 it is when enumerating saved BT devices?

RicardoCarvalho01, 2025-03-15:
@nisargjhaveri , no. Only when I try to build it myself. Changing stuff, or not.
@manio it is at bluetooth's initAdapter.

nisargjhaveri, 2025-03-15:
> @nisargjhaveri , no. Only when I try to build it myself. With or without applying changes.

Curious why there would be a difference just because building locally. Are you using the docker image? Any other local change you can think that can affect this?

In any case, would be possible to wrap initAdapter in try/catch and see if that can give any hints to debug? Though a single uncaught exception should not result in segfault.

Admittedly, our error handling is poor, especially in bluetooth handling. Hope we can fix soon!

RicardoCarvalho01, 2025-03-17:
No, I'm using my wsl machine. I've tried both building everything from source and going through my process of getting each "component".
I did a try/catch around the GetManagedObjects and all it's involved objects but could not get anything out of it