nisargjhaveri/WirelessAndroidAutoDongle #248: Displaying the Android Auto interface on an HDMI display

MrBurritooo, 2025-03-14:
This project has worked flawlessly for the times I've used it on my car, and I had a random thought of thinking maybe the same Android Auto interface would be able to display on a small HDMI display ( Yes I've tried connecting the Pi to a display and it displays the services that are running and it's for debugging purposes, and a small screen with a low resolution would not be fitting). So I was wondering if this were at all possible and if it were, it could also be mounted on bikes and minimize the risk of getting our phones damaged when it's mounted.

manio, 2025-03-14:
You want the same raspberry pi for emulating HUD and connecting your phone?
You may be interested in projects that are able to do it, maybe this?
https://github.com/KreAch3R/aa-proxy-oap

MrBurritooo, 2025-03-14:
Yes, that is what the short version is. I'll check out the project and give my thoughts about it. Thanks for the help!

closed 2025-03-14 by MrBurritooo

MrBurritooo, 2025-03-14:
I've checked out the project, and for it to operate, I need OpenAuto Pro which according to my knowledge is now paid and since a new company has bought, it has almost become obsolete with bugs everywhere. Moreover I have a Pi 3A+ which is not even sufficient for the latest builds and I would have to downgrade the version of Android Auto on my phone cause of compatibility issues. If you have any ideas or suggestions, do let me know. Thanks!

nisargjhaveri, 2025-03-14:
With some tinkering, you might be able to use DHU from Android sdk directly in this case?

I'm really not sure how well it handles real-life scenarios (e.g. phone calls?) though and if there are licensing issues. Works well for limited testing.

MrBurritooo, 2025-03-14:
Might be possible, but I'm not so well versed in Android coding as for the present. But I'll look into your idea. Thanks for guiding me!