nisargjhaveri/WirelessAndroidAutoDongle #250: GPIO enable

nnciprian, 2025-03-21:
Hello,

Would it be feasible to enable GPIO pins? My goal is, while using the RPI Zero 2W for wireless AA, to also read data from a MP280 sensor (and maybe 1 or two other sensors) and show the output via HDMI (my car has HDMI input). HDMI output works and it shows the command line no problem, i guess i could create a script to run at startup for this.
What I am not aware of, is how to enable this with minimal impact over main functionality

manio, 2025-03-21:
Hi,
... so you probably want to make additional use case for the Pi, something similar as I did (in my case for opening garage door). I am using one GPIO input for button and some other for communication with LoRa device. Anyway - @nisargjhaveri should decide, but I think this as a whole is rather out of a scope of this specific AA project. In my case I just created a [separated project](https://github.com/manio/lora-gw-rs) and running it simultaneously on the same Pi.

nnciprian, 2025-03-21:
@manio Yes, ideally i would not want to use an extra device as long as rpi can accept digital input. I guess I will need to add libraries for GPIO and compile a separate script/program for sensor data.
I will have a look on your fork and try to get something form there, thank you for your input

manio, 2025-03-21:
Btw: I don't know if your unit suffer the same "problem" but recently I was fixing the issue that on similar unit as yours (XAV-AX5500) it was necessary to push the complete frames only to the HU (https://github.com/manio/aa-proxy-rs/issues/18#issuecomment-2708244836). The SONY seems to be fastidious in this regard.

Btw2: Is your MP280 similar to BMP280? IIRC I had somewhere some example rust code to read this sensor ;)

nnciprian, 2025-03-22:
1. It works good enough on my Sony, i encountered the sound freeze (similar to back in the days when WinXP would BSoD and and sound froze) on a cold morning when i started the car.For the moment i use a shitty cable as this was my only MicroUSB cable around lol;

2. It's actually BMP280, i misstyped initially;

manio, 2025-03-25:
> LE: speaking of optimization I see saw some commits mentioning similar [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/compare/main...joka90:WirelessAndroidAutoDongle:main) where he mentioned compression disable ( i guess it;s about same full-frame send)

No, the compression you mentioning has nothing in common with the full frames. His work is to speed up booting times and the compression is for filesystem.

nnciprian, 2025-03-25:
> Btw: I don't know if your unit suffer the same "problem" but recently I was fixing the issue that on similar unit as yours (XAV-AX5500) it was necessary to push the complete frames only to the HU ([manio/aa-proxy-rs#18 (comment)](https://github.com/manio/aa-proxy-rs/issues/18#issuecomment-2708244836)). The SONY seems to be fastidious in this regard.
 I will give your build a test and report back

Ioniq3, 2025-04-06:
I add this packages to raspberryXXXX_defconfig in order to get the GPIO access in a python programs, bash too. I add samba, iperf, sensors, exfat and git support.

BR2_PACKAGE_E2FSPROGS=y
BR2_PACKAGE_E2FSPROGS_FSCK=y
BR2_PACKAGE_EXFAT_UTILS=y
BR2_PACKAGE_PROVIDES_LIBEGL="rpi-userland"
BR2_PACKAGE_PROVIDES_LIBGLES="rpi-userland"
BR2_PACKAGE_QT5_GL_AVAILABLE=y
BR2_PACKAGE_QT6_GL_SUPPORTS=y
BR2_PACKAGE_PROVIDES_LIBOPENVG="rpi-userland"
BR2_PACKAGE_LM_SENSORS=y
BR2_PACKAGE_LM_SENSORS_SENSORS=y
BR2_PACKAGE_RPI_USERLAND=y
BR2_PACKAGE_PROVIDES_LIBOPENMAX="rpi-userland"
BR2_PACKAGE_PYTHON3=y
BR2_PACKAGE_PYTHON3_PYC_ONLY=y
BR2_PACKAGE_PYTHON3_UNICODEDATA=y
BR2_PACKAGE_PYTHON_GREENLET_ARCH_SUPPORTS=y
BR2_PACKAGE_PYTHON_NUMPY_ARCH_SUPPORTS=y
BR2_PACKAGE_PYTHON_RPI_GPIO=y
BR2_PACKAGE_GNUTLS=y
BR2_PACKAGE_NETTLE=y
BR2_PACKAGE_HAS_LIBEGL=y
BR2_PACKAGE_HAS_LIBGLES=y
BR2_PACKAGE_HAS_LIBOPENVG=y
BR2_PACKAGE_HAS_LIBOPENMAX=y
BR2_PACKAGE_LIBGPIOD=y
BR2_PACKAGE_LIBGPIOD_TOOLS=y
BR2_PACKAGE_BLUEZ5_UTILS_HEADERS=y
BR2_PACKAGE_LIBTIRPC=y
BR2_PACKAGE_CMOCKA=y
BR2_PACKAGE_GMP=y
BR2_PACKAGE_LIBTASN1=y
BR2_PACKAGE_LIBUNISTRING=y
BR2_PACKAGE_POPT=y
BR2_PACKAGE_GESFTPSERVER=y
BR2_PACKAGE_IPERF=y
BR2_PACKAGE_IPERF3=y
BR2_PACKAGE_SAMBA4=y
BR2_PACKAGE_UTIL_LINUX_LIBBLKID=y
BR2_PACKAGE_UTIL_LINUX_LIBUUID=y
BR2_PACKAGE_GIT=y