nisargjhaveri/WirelessAndroidAutoDongle #251: Can we use SD card remaining storage to store music and play from it like a USB pendrive functionality

shrinivas-t, 2025-03-25:
hi @nisargjhaveri , 
Thank you for this wonderful and easy to follow DIY solution for wireless android auto. I'm using this with raspberry pi zero 2 w and it is working fine.

I have installed it on a 4GB SD card and i could see it creates a volume of around 32MB active, primary partition and one more partition of 120MB as healthy, primary partition. I could see remaining all storage is unallocated/ free.
Can we use this storage to store music and it should get work as a USB pendrive which generally cars have option to play music from.
In my car the port where we connect the raspberry pi is the same port where we can connect any USB to play music from so Ideally It should work.
However when I have tried creating a FAT32 partition of free space and added few music files, but it shows no USB connected. 
I would want to know will this work or is there any other way to make use of that storage for simply storing music files and playing them.

nnciprian, 2025-03-25:
If my understanding is correct,  No, because it does not advertise itself as mass storage device towards the head unit

shrinivas-t, 2025-03-26:
thank you. Understood.

closed 2025-03-26 by shrinivas-t