nisargjhaveri/WirelessAndroidAutoDongle #252: WiFi disconnection issue on pi zero 2w when connected to car

gianros2, 2025-03-29:
Hi,
I’m having an issue with my Raspberry Pi Zero 2W when I connect it to my car. When I plug it in, it connects to bluetooth first, then to wifi, but after about 2 seconds, wifi connection drop. The pi keeps trying to reconnect, but it goes in an endless loop of connecting for 2sec and disconnecting.
However, when I use two cables — one for power and one for OTG — everything works fine, and the pi connects without any issues.

I’m not sure if this is related to power supply from the car’s USB port or something else. Has anyone experienced a similar problem or have any ideas on what might be causing this?

Thanks for any help!

manio, 2025-03-29:
Yeah, first try to look here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/211

vikasjamdagni, 2025-06-07:
I was also having frequent disconnection issues with my Pi zero 2W ... tried all the options changing cable using seperate power cable.... but nothing seems to be good help.... In last I installed a fat ass heat sink over the processor using proper heat sink paste and there is a lot of improvements.... I am not saying its 100% fine but disconnections are know less frequent..... I am using setup in my Nexon Pure 2003 model...... trying adding a good HeatSink you may find improvements.

manio, 2025-07-29:
@gianros2 in this case you can close this issue I think.

regarding @vikasjamdagni look here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/190#issuecomment-3127256335