nisargjhaveri/WirelessAndroidAutoDongle #253: Slow startup?

Jolt151, 2025-03-30:
I'm seeing startup times of 30s to 1min+, using a RPi4. Is this typical for others? Happy to pull logs if others are seeing shorter startup times.

For context, wired connections will always take under 10 seconds, and there's a noticeably longer period of "looking for android auto" with this.

manio, 2025-03-30:
Hi,
There's a similar discussion about this here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/197#issuecomment-2523526112
I was able to went down to 7 seconds :)

nisargjhaveri, 2025-03-30:
RPi 4 generally should take ~10-15 seconds to boot up with the default setup as well. If it takes upto 1min to connect, that generally means it involved some retries on the way. If you can get the logs out, it may help.

That said, there are definitely optimizations possible (Thanks @manio for the reference as well). I have been meaning to get a few of those in by default, but haven't had chance to experiment enough and test the stability (and maintainability) of each. PRs welcome :)

Jolt151, 2025-04-01:
I flashed from head and re-setup everything, applying the aforementioned optimizations. Timed it (after the first boot) and it took almost exactly 1.5 minutes to connect. Nothing in the logs sticks out to my untrained eye: https://pastebin.com/tcFteVsJ

Thanks for taking a look!

manio, 2025-04-01:
I think that a problem may be here:
> Jan  1 00:00:05 buildroot user.info aawgd[276]: Trying to connect bluetooth device at path: /org/bluez/hci0/dev_30_BB_7D_B3_44_68
[...]
> Jan  1 00:00:25 buildroot user.info aawgd[276]: Failed to connect device at path: /org/bluez/hci0/dev_30_BB_7D_B3_44_68

The aawgd is trying to connect to this specific bluetooth device and finally it is giving up 20 seconds later. Is it your BT of your phone? And is it in range, BT powered-on?