nisargjhaveri/WirelessAndroidAutoDongle #254: Can't connect via SCP/SFTP

Jolt151, 2025-04-02:
```
➜  ~ /usr/bin/scp ./Downloads/aa-proxy-rs root@10.0.0.1:/usr/bin
root@10.0.0.1's password:
sh: /usr/libexec/sftp-server: not found
```

Jolt151, 2025-04-02:
cc @manio

manio, 2025-04-02:
This issue should be probably in aa-proxy-rs repo?
I am copying it like this:
```
scp -O YOUR_PI_ADDRESS:/var/log/aa-proxy-rs.log .
```

Jolt151, 2025-04-02:
No, this command is trying to upload the aa-proxy-rs daemon to this image, though it seems that there is no SFTP server in the image. Reverse SCP might work as that might use the laptop's SFTP server instead, though I haven't confirmed that yet. FWIW I worked around this by tunneling the data directly through SSH: https://superuser.com/a/84022

manio, 2025-04-02:
Ok... but I meant this github repository is for WirelessAndroidAutoDongle and aawgd, not aa-proxy-rs.

There is [SCP](https://en.wikipedia.org/wiki/Secure_copy_protocol) (copy using ssh) available but you need to pass `-O` when you are using recent linux distro's `scp` command.

From the manual:

>        -O      Use the legacy SCP protocol for file transfers instead of
>                the SFTP protocol.  Forcing the use of the SCP protocol
>                may be necessary for servers that do not implement SFTP,
>                for backwards-compatibility for particular filename
>                wildcard patterns and for expanding paths with a ‘~’
>                prefix for older SFTP servers.

No matter if you are using scp for obtaining the logs or transferring binary to the Pi (other direction).

Jolt151, 2025-04-02:
You nailed it - `-O` is required here. Thanks!

closed 2025-04-02 by Jolt151