nisargjhaveri/WirelessAndroidAutoDongle #256: connection issues  with rasberry pie 2w

Burner2430, 2025-04-06:
Device: Raspberry Pi 2W used as an Android Auto dongle.
Issue: Disconnection problems caused by software issues, potentially related to WiFi or Bluetooth.
Request: New, updated image or software to resolve connection issues, freezes, and shutter lags.

manio, 2025-04-06:
Request: Logs :)

Burner2430, 2025-04-06:
> Request: Logs :)
i cannot cause i did not make the logs for it nor i cannot make or takes logs now!

michaeldice, 2025-04-09:
I might have a similar issue. Navigation and music apps work great! I really love not having to plug my phone in. However, when a phone call comes in, things go haywire, and the phone appears to rapidly switch between the wireless device and a standard Bluetooth connection. I end up having to unplug the AA dongle to use the phone.

Burner2430, 2025-04-10:
> I might have a similar issue. Navigation and music apps work great! I really love not having to plug my phone in. However, when a phone call comes in, things go haywire, and the phone appears to [...]

hmm but an update needs to come

manio, 2025-04-10:
> I might have a similar issue. Navigation and music apps work great! I really love not having to plug my phone in. However, when a phone call comes in, things go haywire, and the phone appears to [...]

I don't have such an issue - my phone calls are handled properly. Connection is between your phone and a headunit via bluetooth. Make sure that your phone is correctly paired with your HeadUnit.

snaveenkumareee, 2025-04-30:
yes my pie 2w is dropping connection very frequently on the latest release. the music breaks every now and then..  opening any second app drops the connection. I can share the logs if required.. but pls share me the steps on how to peep in

manio, 2025-04-30:
The steps are [here](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#getting-logs).

snaveenkumareee, 2025-05-09:
If the logs do not persist after restart, then it is difficult to capture them in our usecase

manio, 2025-05-09:
If you want to persist than I think you'd need to add a saving the log to some file, which is in third `rw` partition of the SD card, you can do it in [this line](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/blob/main/aa_wireless_dongle/board/common/rootfs_overlay/etc/init.d/S93aawgd#L20) of the startup script.
In [my case](https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#usage) I have a parameter for this, but for aawgd according to [man page](https://man7.org/linux/man-pages/man8/start-stop-daemon.8.html) you can use `-O, --output` parameter to specify the output file - and you should have the logs on persistent partition, but keep in mind that rebooting/starting the pi might overwrite/empty this file, but not sure about it though.

Scg19, 2025-05-19:
`#!/bin/sh
#
# Start aawgd daemon
#

DAEMON="aawgd"
PIDFILE="/var/run/$DAEMON.pid"
LOGFILE="/var/log/$DAEMON.log"

RETVAL=0

case "$1" in
	start)
		if [ -z "$AAWG_CONF_SOURCED" ]; then
			echo "Source /etc/aawgd.conf"
			source /etc/aawgd.conf
		fi

		printf "Starting $DAEMON: "
		start-stop-daemon -S -b -q -m -p "$PIDFILE" -x "/usr/bin/$DAEMON" -- -O "$LOGFILE"
		RETVAL=$?
		[ $RETVAL = 0 ] && echo "OK" || echo "FAIL"
		;;
	stop)
		printf "Stopping $DAEMON: "
		start-stop-daemon -K -q -p "$PIDFILE"
		RETVAL=$?
		[ $RETVAL = 0 ] && echo "OK" || echo "FAIL"
		;;
	restart)
		$0 stop
		$0 start
		;;
	*)
		echo "Usage: $0 {start|stop|restart}"
		;;
esac

exit $RETVAL
`