nisargjhaveri/WirelessAndroidAutoDongle #259: Android Auto frequently connect and disconnect in Hyundai Exter

AkshatAgrawal05, 2025-04-12:
Hello Everyone,

I am using raspberry pi 2w with latest image [raspberrypizero2w-sdcard.img.xz](https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/releases/download/v0.5.0/raspberrypizero2w-sdcard.img.xz)

 The pi keeps trying to reconnect, but it goes in an endless loop of connecting for 2sec and disconnecting.

I also tried two cable method mentioned in https://github.com/nisargjhaveri/WirelessAndroidAutoDongle/issues/211

but still not working... 

attaching logs... 

`Jan  1 00:00:02 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:02 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:02 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2025-03-09 08:49:34 UTC)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1922-927) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sun Mar  9 09:21:32 UTC 2025
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:02 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:02 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:0A:AB:3F vc_mem.mem_base=0x1ec0000
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jan  1 00:00:02 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000018] Switching to timer-based delay loop, resolution 52ns
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000346] Console: colour dummy device 80x30
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000857] printk: console [tty1] enabled
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000909] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.000951] CPU: Testing write buffer coherency: ok
Jan  1 00:00:02 buildroot kern.info kernel: [    0.001009] pid_max: default: 32768 minimum: 301
Jan  1 00:00:02 buildroot kern.info kernel: [    0.001137] LSM: Security Framework initializing
Jan  1 00:00:02 buildroot kern.info kernel: [    0.001260] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.001298] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Jan  1 00:00:02 buildroot kern.info kernel: [    0.002332] cgroup: Disabling memory control group subsystem
Jan  1 00:00:02 buildroot kern.info kernel: [    0.002853] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Jan  1 00:00:02 buildroot kern.info kernel: [    0.003619] cblist_init_generic: Setting adjustable number of callback queues.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.003650] cblist_init_generic: Setting shift to 2 and lim to 1.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.003787] Setting up static identity map for 0x100000 - 0x10003c
Jan  1 00:00:02 buildroot kern.info kernel: [    0.003921] rcu: Hierarchical SRCU implementation.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.003942] rcu: 	Max phase no-delay instances is 1000.
Jan  1 00:00:02 buildroot kern.info kernel: [    0.004580] smp: Bringing up secondary CPUs ...
Jan  1 00:00:02 buildroot kern.info kernel: [    0.005376] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Jan  1 00:00:02 buildroot kern.info kernel: [    0.006182] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 474 lines cut ...]
Jan  1 00:04:24 buildroot user.info aawgd[269]: HSP HS NewConnection
Jan  1 00:04:24 buildroot user.info aawgd[269]: Path: /org/bluez/hci0/dev_AC_D6_18_7B_F6_BF, fd: 29
Jan  1 00:04:24 buildroot user.info aawgd[269]: Bluetooth connected to the device
Jan  1 00:04:24 buildroot daemon.info hostapd: wlan0: STA c0:b6:f9:98:38:bb IEEE 802.11: associated
Jan  1 00:04:24 buildroot daemon.info hostapd: wlan0: STA c0:b6:f9:98:38:bb WPA: pairwise key handshake completed (RSN)
Jan  1 00:04:24 buildroot daemon.info dnsmasq-dhcp[234]: DHCPREQUEST(wlan0) 10.0.0.7 c0:b6:f9:98:38:bb 
Jan  1 00:04:24 buildroot daemon.info dnsmasq-dhcp[234]: DHCPACK(wlan0) 10.0.0.7 c0:b6:f9:98:38:bb Akshat
Jan  1 00:04:24 buildroot user.info aawgd[269]: AA Wireless NewConnection
Jan  1 00:04:24 buildroot user.info aawgd[269]: Path: /org/bluez/hci0/dev_AC_D6_18_7B_F6_BF, fd: 30
Jan  1 00:04:24 buildroot user.info aawgd[269]: Sending WifiStartRequest (ip: 10.0.0.1, port: 5288)
Jan  1 00:04:24 buildroot user.info aawgd[269]: Sent WifiStartRequest, messageId: 1, wrote 17 bytes
Jan  1 00:04:24 buildroot user.info aawgd[269]: Read WifiInfoRequest. length: 0, messageId: 2
[... cut ...]
Jan  1 00:04:25 buildroot kern.info kernel: [  265.974618] dwc2 3f980000.usb: bound driver configfs-gadget.default
Jan  1 00:04:26 buildroot daemon.err bluetoothd[170]: src/profile.c:ext_io_disconnected() Unable to get io data for HSP HS: getpeername: Transport endpoint is not connected (107)
Jan  1 00:04:26 buildroot daemon.err bluetoothd[170]: src/profile.c:ext_io_disconnected() Unable to get io data for AA Wireless: getpeername: Transport endpoint is not connected (107)
Jan  1 00:04:26 buildroot user.info aawgd[269]: Bluetooth adapter was powered off
`

drniraj666, 2025-04-16:
Same problem with my Hyundai Grand i10 Nios.

Ioniq3, 2025-04-16:
try this changes in aa_wireless_dongle/package/aawg/src/bluetoothHandler.cpp , but you have to build again. Thanks to @manio or try the aa-proxy-rs replacement from him, it works great.

sed -i '2i #include <unistd.h>' aa_wireless_dongle/package/aawg/src/bluetoothHandler.cpp
sed -i '103i     int i = 10;' aa_wireless_dongle/package/aawg/src/bluetoothHandler.cpp
sed -i '104i      while (i-- && !m_adapter->powered->value()) {sleep(1);}' aa_wireless_dongle/package/aawg/src/bluetoothHandler.cpp

AkshatAgrawal05, 2025-04-18:
Hi @Ioniq3 , 

Tried aa-proxy-rs as well facing same issue.... thanks to @manio... he is constantly supporting and helping me to debug this

I am not sure, issue is with Hyudai's head unit or something else...