nisargjhaveri/WirelessAndroidAutoDongle #260: Having to unlock the phone for AntennaPod to work with Android Auto Wireless Dongle only

cprados, 2025-04-14:
Hi,

Have anyone noticed something similar? I Use this app for listening podcasts. Only when using it through this dongle I have noticed it requires me to unlock the phone for listing the podcasts episodes or listing subscribed podcasts.

I have been testing it both with AA through cable and through the dongle several weeks and I couldn't reproduce the issue through cable. Through AA wireless adapter it happens easily almost every time I use that app.

I'm not saying it is this project fault's nor Antenapod fault's, I'm only wondering if someone noticed similar issues with that app or with other through the dongle.

Best regards.

manio, 2025-04-14:
It is really weird because the dongle is doing transparent proxy. The only difference is probably - phone is detecting "Wireless" AndroidAuto mode and do something differently. For sure the transfer might be different depending on media, eg 2.4/5GHz connection. I know that some phones (can't remind exactly but AFAIR it was pixels or motorola) has a bug which prevent using AndroidAuto before prior unlocking phone. But not sure if it is the same issue.

What I can suggest is maybe trying new `video-in-motion` feature in [aa-proxy-rs](https://github.com/manio/aa-proxy-rs), not sure though if this help....

nisargjhaveri, 2025-04-19:
There is a "Start Android Auto while locked" in Android Auto settings. If it works with wired, not sure if this cab be a problem, just to make sure.

cprados, 2025-04-24:
Yes, Start Android Auto While Locked has allways been enabled so this doesn't seem to be the issue.

I also tend to think that it is phone's fault doing something differently when it is detecting wireless AA.

cprados, 2025-04-30:
> It is really weird because the dongle is doing transparent proxy. The only difference is probably - phone is detecting "Wireless" AndroidAuto mode and do something differently. For sure the transfer [...]

@manio I've been testing lastest version of your project for some days now and it's going fine so far. Why do you video-in-motion option could help in this specific case?

manio, 2025-04-30:
> Why do you think video-in-motion option could help in this specific case?

It was only just a guess basing on your input about your podcasts application. This option is faking applications that the car is not moving allowing doing actions which are allowed in this case, like e.g. watching videos, etc.