nisargjhaveri/WirelessAndroidAutoDongle #261: Feature Suggestion: ODB2 Data

juggie, 2025-04-18:
Hey!

There are multiple cars that feed battery stats over usb or android wireless android auto that's built in to google maps can be aware of say battery SOC, etc and make battery predictions on routes.

However, this is rare, or at least rarer than it should be.  One solution to this is to use an ODB2 Dongle + A better route planner (android auto app) which reads from ODB2.

What would be a nice feature is if it was possible to inject the battery stats from odb2 into the android auto data that's passing through.  This would add EV routing to any car that supports android auto even if the manufacturer won't update their own system.  Let me know what you think! I would sponsor a bit for this feature for sure.. and I think if advertised many others also would!

manio, 2025-04-19:
+1000 for this :)
That's been a thorn in the side lately despite our "campaign":
https://www.reddit.com/r/AndroidAuto/comments/1jmh234/searching_for_people_using_ev_features_of_aa/
https://www.reddit.com/r/MachE/comments/1jpj2h6/searching_for_people_using_ev_features_of/
https://www.macheforum.com/site/threads/please-help-figuring-out-ev-features-for-androidauto.44456/

The main problem is that it's hard to find someone who:
a) is willing to help and doesn't fear connecting Raspberry to its car USB port
b) have some basic geek knowledge - raspberry Pi owner/able to write SD card image, etc

Maybe the situation will be better over time when more cars with EV routing features will appear (besides of Ford MachE)...

ps. We are trying hard to figure this EV features out and I can see you've joined our discord already so you're familiar with the problems :)

richiethom, 2025-07-09:
[aa-proxy-rs](https://github.com/manio/aa-proxy-rs) is making progress here, and it is designed as a drop-in for WirelessAndroidAutoDongle. It has a man-in-the-middle feature that is [able to inject the EV SoC into the Android Auto 'feed'](https://github.com/manio/aa-proxy-rs) meaning that the Google Maps EV functions appear on cars that don't support it. I have no idea how mature aa-proxy-rs is in this respect, but the dev has shown screenshots of it working on his old EV.