nisargjhaveri/WirelessAndroidAutoDongle #262: 2018 Jeep JLU Sahara

sharmay, 2025-04-21:
I was setting up Pi Zero 2 W and ran into a constant reboot issues.
Bluetooth visible
Connect phone to Bluetooth
Popup message "Unable to connect to <Bluetooth Device Address>", this happens due to PI Zero reboots

Q. Any settings to make the logs persistent?
It is not easy to retrieve logs due to constant reboot

manio, 2025-04-21:
If you have reboots than I think it may be some kind of power issue. The logs can be easily extracted but only when the Pi is not rebooting. I am personally also using Pi Zero 2 W.

sharmay, 2025-04-22:
My other car is 2023 Kia Car running 0.5.0 with Pixel 7Pro. I tried setting up another phone Samsung S22 Ultra which failed and during next drive it gets in constant reboot cycle.

I will try with secondary power.

sharmay, 2025-05-06:
> If you have reboots than I think it may be some kind of power issue. The logs can be easily extracted but only when the Pi is not rebooting. I am personally also using Pi Zero 2 W.

I tested with secondary power with a brand new image on sdcard but same reboot loop.

manio, 2025-05-07:
Hmm.... this still seems for me like a power/hardware problem... maybe as the last resort try to connect some small HDMI display to the HDMI port to visually see what is going on...

sharmay, 2025-07-15:
> 
> OK, everything's clear... it's a hardware issue... check your power, maybe you can power it from external source using split usb cable...?

Split cable or add an another cable to power only micro usb port?

manio, 2025-07-15:
OK... your logs is not telling us anything new. It is only confirming that you have a hardware issue. It is not normal that raspberry is restarting. And there may be multiple reasons for this: weak power from USB port, overheating, SD card problems, etc...

> Split cable or add an another cable to "power only micro usb port"?

This kind of thing for example:
[image]
I mean the cable for which you split data and power. And you can get the power from cigarette 12V charger and only data to the headunit.

Do you have problems only when connected to the car? What if you connect it via charger but eg in your home... is it working reliable, not rebooting?

sharmay, 2025-07-15:
I will order this cable,  mean while:

[image]

I powered it using either of micro usb when connected to a powerbank, no crash.

I powered it using the right side micro usb and connected to a Jeep, no crash but as soon as I also connect the left side micro usb to Jeep, crash starts.

I powered it using the right side micro usb connected to a powerbank, no crash but as soon as I connect the left micro usb to Jeep, crash starts.

Q. Since PI has 2 micro USB Power and Data, can't I use both?  Rigt side to power and left side for data?

manio, 2025-07-16:
Hmmm... it makes sense, and in this case probably you don't need this cable :)
The cable was mentioned in regards connecting Pi5 where it is unable to grab power the other way...

I heard about one case though when someone broke their USB port in HU when "playing" with this USB and power things. But it was probably using USB hub, right, @Ioniq3 ?

sharmay, 2025-07-21:
Unfortunately, Pi still crashed when I connected both power (power only cable) and data usb ports.

Ioniq3, 2025-07-21:
> Hmmm... it makes sense, and in this case probably you don't need this cable :) The cable was mentioned in regards connecting Pi5 where it is unable to grab power the other way...
>
> I heard about one [...]

Raspberry pi 0w, 2w and 3a doesn't need to power by power only microusb, all these board consume less than 500mA of 5V and can be power only by the data+power usb connected to the car.

With hubs, and if you want to power it you have to be carefull, the best is to isolate +5V line from car to the hub.

kurtj, 2025-07-29:
Here are the logs I grabbed from when trying connection mode 2 ( no reboot loop, but no successful launch of wireless android auto)

```
# cat /var/log/
dbus/           dnsmasq.leases  hostapd         messages
# cat /var/log/messages
Jan  1 00:00:03 buildroot syslog.info syslogd started: BusyBox v1.36.1
Jan  1 00:00:03 buildroot user.info console: Starting syslogd: OK
Jan  1 00:00:03 buildroot kern.notice kernel: klogd started: BusyBox v1.36.1 (2025-03-09 08:49:34 UTC)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Linux version 6.1.61-v7 (runner@fv-az1922-927) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sun Mar  9 09:21:32 UTC 2025
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] random: crng init done
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Zone ranges:
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   Normal   empty
Jan  1 00:00:03 buildroot user.info console: Starting klogd: OK
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Movable zone start for each node
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Early memory node ranges
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
Jan  1 00:00:03 buildroot kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 104545
Jan  1 00:00:03 buildroot kern.notice kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:34:FF:24 vc_mem.mem_base=0x1ec0000
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] Memory: 337224K/421888K available (9216K kernel code, 718K rwdata, 2316K rodata, 1024K init, 520K bss, 19128K reserved, 65536K cma-reserved)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000]      Tracing variant of Tasks RCU enabled.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000019] Switching to timer-based delay loop, resolution 52ns
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000350] Console: colour dummy device 80x30
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000862] printk: console [tty1] enabled
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000914] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Jan  1 00:00:03 buildroot kern.info kernel: [    0.000955] CPU: Testing write buffer coherency: ok
[... cut ...]
Jan  1 00:02:31 buildroot authpriv.info dropbear[278]: Generated hostkey /etc/dropbear/dropbear_ed25519_host_key, fingerprint is SHA256:IxGGHFMMGYU9cNXyyRd5a3gsGRTL5t53G1b4DtFIhc0
Jan  1 00:02:32 buildroot authpriv.info dropbear[279]: Child connection from 10.0.0.2:34088
Jan  1 00:02:33 buildroot authpriv.notice dropbear[279]: Password auth succeeded for 'root' from 10.0.0.2:34088
Jan  1 00:02:36 buildroot authpriv.info dropbear[278]: Exit before auth from <10.0.0.2:34084>: Exited normally
#
```

kurtj, 2025-07-30:
Tried the version you linked to but no joy.  Is there a config fill I should have modified to disable legacy mode?  I didn't'see it so assumed that was an image with legacy mode already disabled.  The version you linked to also did NOT work in the tacoma.

manio, 2025-07-31:
> I'm having trouble connecting to the WiFi to make the changes - whats the wifi password?

Depend what image you flashed (aa-proxy-rs?) and if you enabled WiFi password in aawgd.conf