nisargjhaveri/WirelessAndroidAutoDongle #264: Wifi and Bluetooth question

Siddharth-Stallion, 2025-04-22:
I have a Nexon 2021. When I connected my raspberry pi zero w, it says connected in the Bluetooth (the name is WirelessAADongle-731ec0). However, I am not able to see the AAWiredDongle option in the wifi settings. Maybe this issue has been solved before, or maybe I missed out somehhthig. Could you please help me out?

manio, 2025-04-22:
Hi, the steps are the following:
1. It's depending on strategy but we can say that you are connecting to a dongle bluetooth for the first time
2. The dongle is starting connection via usb with HeadUnit
3. Your phone is automatically connecting to specified WiFi to the Raspberry (you don't need that info as it is passed via BT channel)
4. Your phone is automatically connecting to car HeadUnit Bluetooth for audio calls

Siddharth-Stallion, 2025-04-22:
Ok, so once the Bluetooth is connected for calls, the wifi should get automatically connected right?

manio, 2025-04-22:
Yeah, if you want some technical explanation you can read here:
https://github.com/manio/aa-proxy-rs?tab=readme-ov-file#how-it-works-technical

Siddharth-Stallion, 2025-04-22:
The issue is, once the Bluetooth gets connected, I'm not able to see the option to connect wifi

Siddharth-Stallion, 2025-04-22:
Yes, I want to connect to the AAWirlessDongle wifi network. Unfortunately, I am not able to see an option to do that

manio, 2025-04-22:
Can I ask why you want to connect to it? To obtain the logs?

Siddharth-Stallion, 2025-04-22:
My Android phone

Siddharth-Stallion, 2025-04-22:
Sorry, the steps ask me to connect Bluetooth and then connect to the AAWirlessDongle via my phone's wifi settings

manio, 2025-04-22:
No it is not working like this. You connect to the dongle via Bluetooth for the first time, then the magic happens and you don't need to do anything more!
Your phone should connect automatically (without your intervention) to the dongle via WiFi for Wireless AndroidAuto.

Siddharth-Stallion, 2025-04-22:
Then why doesn't my car recognise any AA device connected? I have followed the steps correctly.

manio, 2025-04-22:
Steps I mentioned was only to provide overview what's going on/how it is working - keep in mind that your active part is only in point one - when you are connecting to the dongle - rest is done automatically.

It may be various reasons for this. I think that logs would be useful here to trying to tell what is the problem...
Also: look here: https://github.com/nisargjhaveri/WirelessAndroidAutoDongle?tab=readme-ov-file#troubleshoot