nisargjhaveri/WirelessAndroidAutoDongle #265: Workable cables?

markleigh, 2025-04-23:
Does anyone have a link to a micro USB cable spec/brand/type/model that seems to work flawlessly and dependably for this application? I'm having issues, but I suspect it's not a software issue but rather a cable issue.
Thanks in advance!

manio, 2025-04-23:
Heh... I am using this old nokia cable. I like it because it is short, so perfectly for car application:
[image]

What is your issue exactly?

baudhayan, 2025-05-10:
I too use this Nokia Cable. @markleigh - it needs to be data transfer cable rather than a charging cable.

itsjustasemicolon, 2025-06-02:
Any cable that has both power and data cables (Usually cables that came with older phones in the box will work just fine)